leggedrobotics/darknet_ros #125: hi,how can I run the test for detecting an image?

Tkzccsk, 2018-10-20:


JiyongOh, 2018-10-30:
I have tested this darknet_ros package using PointGrey camera Grasshopper3.
After building the package, type the following command.
roslaunch darknet_ros yolo_v3.launch

If it does not work and your terminal put the message "Wait for image" repeatly,
check topic name in the file of /darknet_ros/darknet_ros/config/ros.yaml.
I changed the topic name of 'camera_reading' from '/camera/rgb/image_raw' to '/camera/image_raw'.

Good luck.

Tyagi-Ramakrishnan, 2019-03-06:
Hi,

That trick did not work for me. It still puts out waiting for image repeatedly.

Tyagi

liqsuq, 2019-06-10:
I have same problem.

I want to receive other device's ros topics.
I set some envs and looked "rostopic list" before launch darknet_ros.

$ rostopic list
:
/v4l/camera/camera_info
/v4l/camera/image_raw
/v4l/camera/image_raw/compressed
/v4l/camera/image_raw/compressed/parameter_descriptions
/v4l/camera/image_raw/compressed/parameter_updates
/v4l/camera/image_raw/compressedDepth
/v4l/camera/image_raw/compressedDepth/parameter_descriptions
/v4l/camera/image_raw/compressedDepth/parameter_updates
/v4l/camera/image_raw/theora
/v4l/camera/image_raw/theora/parameter_descriptions
/v4l/camera/image_raw/theora/parameter_updates

Then I changed the topic name of 'camera_reading' from '/camera/rgb/image_raw' to '/v4l/camera/image_raw'.
But It still wait for image repeatedly.

Would you tell me some ideas?

Thanks.

liqsuq, 2019-06-10:
Above problem was self-solved.

Publishing camera's settings by other was wrong configured.
(Not setting ROS_HOSTNAME)
 
However, I got to know the timing to use "rqt_image_view" due to him.

Thanks.

Fenix0817, 2019-08-09:
Hello everyone,

I resolved this problem changing the code line 158-159 of YoloObjectDetector.cpp (At Augost 2019) with this line:

imageSubscriber_ = imageTransport_.subscribe(cameraTopicName, cameraQueueSize,
                                               &YoloObjectDetector::cameraCallback, this, image_transport::TransportHints("compressed"));

The secret, in my opinion, it was had to add image_transport::TransportHints("compressed") in the end.

Regards.