leggedrobotics/darknet_ros #127: catkin_make error

shaarkrat2, 2018-10-26:
hi

when i use this command
`catkin_make -DCMAKE_BUILD_TYPE=Release`

i get this error

```
CMake Error at darknet_ros-master/darknet_ros/CMakeLists.txt:160 (add_library):
  Cannot find source file:

    DARKNET_PATH-NOTFOUND/src/activation_layer.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx

CMake Error: CMake can not determine linker language for target: darknet_ros_lib
CMake Error: Cannot determine link language for target "darknet_ros_lib".
-- Generating done
-- Build files have been written to: /home/arnaud/yoloros_ws/build
Invoking "cmake" failed

```

i'm under ubuntu 16.04, with ros kinetic

Thanks

shaarkrat2, 2018-10-26:
just install the packages with 
https://github.com/leggedrobotics/darknet_ros

closed 2018-10-26 by shaarkrat2

JayHsiang, 2019-04-04:
I have encountered this same problem, so what is the difference between this and the master branch?
```
CMake Error at /usr/share/cmake-3.5/Modules/FindCUDA.cmake:1649 (add_library):
  Cannot find source file:

    DARKNET_PATH-NOTFOUND/src/activation_layer.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
Call Stack (most recent call first):
  darknet_ros/darknet_ros/CMakeLists.txt:97 (cuda_add_library)

-- Generating done
-- Build files have been written to: /home/lab/ros/build
```
i'm also under ubuntu 16.04, with ros kinetic

Thanks

ChunJyeBehBeh, 2019-04-11:
Hello, how you manage to solve it?

JayHsiang, 2019-04-11:
@ChunJyeBehBeh 
just git the packages recursively with `git clone --recursive git@github.com:leggedrobotics/darknet_ros.git`

zulfiz, 2019-06-25:
I am getting the same error. Here is the screenshot of my error
[image]

I already cloned the library recursively using the command:
git clone --recursive git@github.com:leggedrobotics/darknet_ros.git

Could you please help me  @JayHsiang @shaarkrat2 @ChunJyeBehBeh in solving this issue. Thank you!

1329844541, 2019-07-22:
I have the same question.Did you solve it?

JayHsiang, 2019-07-22:
@1329844541 
just git the packages recursively with
git clone --recursive git@github.com:leggedrobotics/darknet_ros.git
Remiand that --recursive is necessary.

MccAlfh, 2020-09-22:
Setting the right path of DARKNET also can solve this problem，you can find the CMAKE log likes 

【-- +++ processing catkin package: 'darknet_ros'
-- ==> add_subdirectory(darknet_ros/darknet_ros)
-- Darknet path dir = /home/alfh/catkin_ws/src/darknet_ros-master/darknet】

Your darknet path should be corresponding to “-- Darknet path dir = /home/alfh/catkin_ws/src/darknet_ros-master/darknet”
may this be helpful！

everardo07, 2021-07-08:
Hello 

I have the same question? How do you solve the problem , cause I tried the solutions showed in the previous comments but nothing works.

mf093087, 2021-12-07:
> Hello
> 
> I have the same question? How do you solve the problem , cause I tried the solutions showed in the previous comments but nothing works.

Me too, nothing works

SkyWxxk, 2022-03-10:
I met the  same problem, because I downloaded it  by typing
`git clone https://github.com/leggedrobotics/darknet_ros.git`
Later I added the` --recursive ` like 
`git clone --recursive https://github.com/leggedrobotics/darknet_ros.git`
Then compiled successfully

bibinthomas007, 2023-09-22:
If you haven't used --recursive flag while cloning the repository, do the below steps.
Navigate to the darknet_ros directory inside your ws
cd ~/catkin_ws/src/darknet_ros
Initialize and update the submodules:
git submodule update --init --recursive
next do 
catkin_make once
then, 
catkin_make -DCMAKE_BUILD_TYPE=Release