leggedrobotics/darknet_ros #129: Build time is approaching 30 minutes

benjaminabruzzo, 2018-10-30:
Following the installation instructions:
cd ~/ros/src
git clone --recursive git@github.com:leggedrobotics/darknet_ros.git
cd ~/ros
catkin build darknet_ros -DCMAKE_BUILD_TYPE=Release

Build time is approaching 30 minutes.  This seems unusual.

benjaminabruzzo, 2018-10-30:
[build] Package table is up to date.                                                                                                                                                                                                         
Starting  >>> darknet_ros_msgs                                                                                                                                                                                                               
Finished  <<< darknet_ros_msgs                [ 1.3 seconds ]                                                                                                                                                                                
Starting  >>> darknet_ros                                                                                                                                                                                                                    
No handlers could be found for logger "trollius"ed] [darknet_ros:check - 1.2]                                                                                                                                                                
^C[build] Interrupted by user!                                                                                                                                                                                                                 
[build] Summary: 1 of 2 packages succeeded.                                                                                                                                                                                                  
[build]   Ignored:   25 packages were skipped or are blacklisted.                                                                                                                                                                            
[build]   Warnings:  None.                                                                                                                                                                                                                   
[build]   Abandoned: 1 packages were abandoned.                                                                                                                                                                                              
[build]   Failed:    None.                                                                                                                                                                                                                   
[build] Runtime: 55 minutes and 2.3 seconds total.

benjaminabruzzo, 2018-10-30:
Using catkin build darknet_ros -DCMAKE_BUILD_TYPE=Release on gnome 16.04 with ros kinetic, the cuda finding part of the cmaklists was failing.  I forked this repo and commented out the cuda detection and it seems like it installed properly.

closed 2018-10-30 by benjaminabruzzo

cardboardcode, 2019-09-20:
Hi there.

Thanks @benjaminabruzzo  for recording and resolving this issue.

# [Possible Solutions]

If you are still having this particular issue where catkin building/making **darknet_ros** is taking forever, try the following:
1. **Ensure** when downloading the [**darknet_ros** github repo](https://github.com/leggedrobotics/darknet_ros), use the following command, as stated on its readme.md

`git clone --recursive git@github.com:leggedrobotics/darknet_ros.git`

2. **Clean** out your `catkin workspace` and build the **darknet_ros** ROS package **FIRST**. The sequence of commands below does exactly that.

>  `cd ~/catkin_ws`
> `catkin clean --y`
> `catkin build -j8 darknet_ros -DCMAKE_BUILD_TYPE=Release`
> `source devel/setup.bash`

3. **Comment** out the cuda detection in `
darknet_ros/darknet_ros/CMakeLists.txt `, as suggested earlier.

> To do so, simply replace the `CMakeLists.txt` under `darknet_ros/darknet_ros` with this specific alternative `CMakeLists.txt` from [**benjaminabruzzo**'s forked repo](https://github.com/benjaminabruzzo/darknet_ros/blob/master/darknet_ros/CMakeLists.txt).

Hi @benjaminabruzzo, please correct this if I am wrong. 

For anyone facing this problem in the future, hope this helps.