leggedrobotics/darknet_ros #131: darknet_ros roslaunch error

JonasPruessing, 2018-11-15:
Hello,
I am new here, can someone help with the error below?

ros_ws/src/darknet_ros/darknet_ros/launch$ roslaunch darknet_ros_gdb.launch
... logging to /home/jonas/.ros/log/ec0cd81e-e860-11e8-b899-901b0e877f68/roslaunch-lab-pc3-29819.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Resource not found: darknet_ros
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

Haya-Alsharif, 2018-11-18:
Add the following to your ~/.bashrc: `export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:~/catkin_ws/src/darknet_ros/darknet_ros`

JonasPruessing, 2018-11-18:
Thank you, problem solved.

JonasPruessing, 2018-11-18:
Problem solved

closed 2018-11-18 by JonasPruessing