leggedrobotics/darknet_ros #133: Error when building with OpenCV 3.4

yangautumn, 2018-12-07:
Hi,
When building with **OpenCV 3.4.1**, I have the following error. Anyone got the same error? I am stucked here. Or I need to install **OpenCV 3.3** since it built successfuly on my other computer. Thanks!

```
Errors     << darknet_ros:make /home/robot/husky_ws/logs/darknet_ros/build.make.008.log                                                                                          
In file included from /usr/local/include/opencv2/core/core_c.h:48:0,
                 from /usr/local/include/opencv2/highgui/highgui_c.h:45,
                 from /home/robot/husky_ws/src/darknet_ros/darknet/include/darknet.h:25,
                 from /home/robot/husky_ws/src/darknet_ros/darknet/src/box.h:3,
                 from /home/robot/husky_ws/src/darknet_ros/darknet/src/image.h:9,
                 from /home/robot/husky_ws/src/darknet_ros/darknet_ros/include/darknet_ros/image_interface.h:12,
                 from /home/robot/husky_ws/src/darknet_ros/darknet_ros/src/image_interface.c:9:
/usr/local/include/opencv2/core/types_c.h: In function ‘cvIplImage’:
/usr/local/include/opencv2/core/types_c.h:370:12: error: incompatible types when returning type ‘int’ but ‘IplImage {aka struct _IplImage}’ was expected
     return _IplImage();
            ^~~~~~~~~~~
make[2]: *** [CMakeFiles/darknet_ros_lib.dir/build.make:1930: CMakeFiles/darknet_ros_lib.dir/src/image_interface.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:624: CMakeFiles/darknet_ros_lib.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

psicobloc, 2019-02-20:
could you fix it? what was it?

yangautumn, 2019-02-20:
> could you fix it? what was it?

@psicobloc yes, it was fixed. The problem is not the OpenCV version, but I had multiple OpenCV versions installed and the installation got confused.

widedh, 2019-03-22:
did you delete one version of the OpenCV versions or you just modify the config?

yangautumn, 2019-03-26:
> did you delete one version of the OpenCV versions or you just modify the config?

@widedh Sorry for the delay. I think both would work, but I deleted the other version.

widedh, 2019-03-27:
@yangautumn thank you for your answer. I will be greatful if you can tell me how did you delete one version because I can't find a solution.

karolmajek, 2019-03-29:
With OpenCV 3.4.0 it should work
bug was introduced in 3.4.1 in C API:
https://github.com/opencv/opencv/issues/10963
I'm building 3.4.0 now

jksLearning, 2019-04-10:
> @yangautumn thank you for your answer. I will be greatful if you can tell me how did you delete one version because I can't find a solution.

Did you figure out how to delete the old version or update the config? I am also stuck on this problem! Thanks you!

haosen9527, 2020-01-14:
yes, you should edit CMakelists.txt
``` CMAKE
find_package(OpenCV 3.4.0 REQUIRED)
```
Wrong Reason darknet vision git checkout 508381b

romain130492, 2024-09-26:
> ```cmake
> (OpenCV 3.4.0
> ```

same same bug..