leggedrobotics/darknet_ros #138: Tiny Yolo V3 version

pnambiar, 2019-01-02:
Does tiny yolo V3 version work with tis wrapper?

tooskoolforkool, 2019-01-09:
Yes

- Create a yolov3_tiny.yaml file in cfg directory and make sure it has the correct information
- Download yolov3_tiny.weights and yolov3_tiny.cfg to the neural_network_config file
- Create a launch file to run yolov3_tiny

mbjelonic, 2020-01-14:
Thanks for the answer!

closed 2020-01-14 by mbjelonic

ican9595, 2021-05-06:
@tarek-mahmoud thanks for the answer but dont I need also the pretrained yolov3-tiny.conv file to train my custom object? where can I download that?

tarek-mahmoud, 2021-05-08:
@ican9595 go to https://pjreddie.com/darknet/yolo/ and you will find two hyperlinks per entry. One for the weights and one for the cfg. 

To train the model, I recommend you follow the instructions outlined in this repo: https://github.com/AlexeyAB/darknet

After training from that repo, you can use the model you trained and port it into your ROS nodes using leggedrobotics's darknet_ros repo