leggedrobotics/darknet_ros #139: error with boost

Aaron9477, 2019-01-04:
Hi, guy~
I have a error with boost. I have try boost 1.58 1.69, but the error always exists.
`In file included from /opt/ros/kinetic/include/actionlib/server/handle_tracker_deleter.h:42:0,
                 from /opt/ros/kinetic/include/actionlib/server/action_server.h:51,
                 from /opt/ros/kinetic/include/actionlib/server/simple_action_server.h:42,
                 from /home/zq610/WYZ/workspace/darknet_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:24,
                 from /home/zq610/WYZ/workspace/darknet_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/opt/ros/kinetic/include/actionlib/destruction_guard.h: In member function ‘void actionlib::DestructionGuard::destruct()’:
/opt/ros/kinetic/include/actionlib/destruction_guard.h:62:80: error: no matching function for call to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>::subsecond_duration(float)’
       count_condition_.timed_wait(lock, boost::posix_time::milliseconds(1000.0f));
                                                                                ^
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_config.hpp:16:0,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_system.hpp:13,
                 from /usr/local/include/boost/date_time/posix_time/ptime.hpp:12,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_types.hpp:12,
                 from /usr/local/include/boost/thread/thread_time.hpp:11,
                 from /usr/local/include/boost/thread/detail/platform_time.hpp:11,
                 from /usr/local/include/boost/thread/pthread/condition_variable.hpp:9,
                 from /usr/local/include/boost/thread/condition_variable.hpp:16,
                 from /usr/local/include/boost/thread/condition.hpp:13,
                 from /opt/ros/kinetic/include/actionlib/server/simple_action_server.h:40,
                 from /home/zq610/WYZ/workspace/darknet_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:24,
                 from /home/zq610/WYZ/workspace/darknet_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note: candidate: template<class T> boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<Functor>, void>::type*)
     explicit subsecond_duration(T const& ss,
              ^
/usr/local/include/boost/date_time/time_duration.hpp:285:14: note:   template argument deduction/substitution failed:
/usr/local/include/boost/date_time/time_duration.hpp: In substitution of ‘template<class T> boost::date_time::subsecond_duration<base_duration, frac_of_second>::subsecond_duration(const T&, typename boost::enable_if<boost::is_integral<Functor>, void>::type*) [with T = float]’:
/opt/ros/kinetic/include/actionlib/destruction_guard.h:62:80:   required from here
/usr/local/include/boost/date_time/time_duration.hpp:285:14: error: no type named ‘type’ in ‘struct boost::enable_if<boost::is_integral<float>, void>’
In file included from /usr/local/include/boost/date_time/posix_time/posix_time_config.hpp:16:0,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_system.hpp:13,
                 from /usr/local/include/boost/date_time/posix_time/ptime.hpp:12,
                 from /usr/local/include/boost/date_time/posix_time/posix_time_types.hpp:12,
                 from /usr/local/include/boost/thread/thread_time.hpp:11,
                 from /usr/local/include/boost/thread/detail/platform_time.hpp:11,
                 from /usr/local/include/boost/thread/pthread/condition_variable.hpp:9,
                 from /usr/local/include/boost/thread/condition_variable.hpp:16,
                 from /usr/local/include/boost/thread/condition.hpp:13,
                 from /opt/ros/kinetic/include/actionlib/server/simple_action_server.h:40,
                 from /home/zq610/WYZ/workspace/darknet_ws/src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:24,
                 from /home/zq610/WYZ/workspace/darknet_ws/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:10:
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate: boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>::subsecond_duration(const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>&)
   class BOOST_SYMBOL_VISIBLE subsecond_duration : public base_duration
                              ^
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note:   no known conversion for argument 1 from ‘float’ to ‘const boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>&’
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note: candidate: boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>::subsecond_duration(boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>&&)
/usr/local/include/boost/date_time/time_duration.hpp:270:30: note:   no known conversion for argument 1 from ‘float’ to ‘boost::date_time::subsecond_duration<boost::posix_time::time_duration, 1000l>&&’
`
I have spent nearly one day on it. Do you have any solution?
Looking forward to your reply
Best wishes!

Aaron9477, 2019-01-06:
I find out the reason is that when I change the version of boost, I just delete the build folder. When I also delete three folders under devel, it works.
Sorry to bother you!
Best wished!

closed 2019-01-06 by Aaron9477

dregner, 2020-04-05:
Which version of boost do you used?