leggedrobotics/darknet_ros #140: Run Multiple Darknet ROS Nodes Each on a Different Camera

tooskoolforkool, 2019-01-09:
**TL;DR?**
Is there a way to run multiple darknet_ros nodes each on a different camera?

**Details:**
I have two cameras connected to the same robot. These cameras are pointing to different directions. I would like one camera to detect persons and I want the other camera to detect abstract shapes. Therefore, I trained two different neural networks.

I am using a single ROS master on the robot. How can I run two instances of darknet at the same time?

**What I Tried So Far**
I ran the first instance and it went fine. When I ran the second instance of darknet_ros, the first instance stopped working. (process ended)
_Next_
I tried to change the name of the node in one of the launch files and I was able to run two darknet_ros nodes simultaneously. However, the node with the modified name gets stuck on _Waiting for image_
_Next_
I copied darknet_ros package so that I had two packages (different names) in the same workspace. It failed to build
_Next_
I copied darknet_ros/darknet_ros so that I had darknet_ros/darknet_ros and darknet_ros/darknet_ros_new, and I tried to build, the darknet_ros_new was ignored during building.
_Next_
I went to the C++ code for the nodes and I replaced all occurrences of darknet_ros into darknet_ros_new

hsaleem1, 2020-12-12:
@NicolasBernard456, could you please further guide that how to change the parameters files and which changes are needed?

hsaleem1, 2020-12-12:
> **TL;DR?**
> Is there a way to run multiple darknet_ros nodes each on a different camera?
>
> **Details:**
> I have two cameras connected to the same robot. These cameras are pointing to different [...]

@tarek-mahmoud, are you done with this problem? If so, please guide.

tarek-mahmoud, 2021-01-04:
@hsaleem1, yes I was able to resolve this problem. Please check below for a sample:

[https://gist.github.com/tarek-mahmoud/eaa62fb91cff8ee73c537558952c1917](url)

closed 2021-01-04 by tarek-mahmoud

NidalFayad, 2021-06-10:
Hello, kindly I have raspberry pi 4, darknet on it but I don't know which topic should I subscribe to for the camera to get an image? I'm also connecting my raspberry pi to pixhawk and using mavros for communication.

idochi, 2022-05-12:
I have a similar problem. My robot has also 2 cameras and both should work over 2 launch files, but it doesn't work. I added both of them to the ros.yaml file , but just one is working , the other one is in the "Waiting for image" mode. Can somebody help me. I'm  new to ROS.

ashiqa, 2023-04-03:
> @hsaleem1, yes I was able to resolve this problem. Please check below for a sample:
> 
> ```
> <?xml version="1.0" encoding="utf-8"?>
> 
> <launch>
>   
>   <arg name="launch_prefix" default=""/>
> 
>   
>   <arg name="yolo_weights_path"          default="$(find darknet_ros)/yolo_network_config/weights"/>
>   <arg name="yolo_config_path"           default="$(find darknet_ros)/yolo_network_config/cfg"/>
> 
>   
>   <rosparam command="load" ns="down_looking_camera/darknet_ros" file="$(find darknet_ros)/config/ros.yaml"/>
>   <rosparam command="load" ns="down_looking_camera/darknet_ros" file="$(find darknet_ros)/config/yolov2-tiny.yaml"/>
>   <rosparam command="load" ns="front_looking_camera/darknet_ros" file="$(find darknet_ros)/config/ros_tags.yaml"/>
>   <rosparam command="load" ns="front_looking_camera/darknet_ros" file="$(find darknet_ros)/config/yolov3-tiny.yaml"/>
> 
>   
>   <node  ns="down_looking_camera" pkg="darknet_ros" type="darknet_ros" name="darknet_ros" output="screen" launch-prefix="$(arg launch_prefix)">
>     <param name="weights_path"                              value="$(arg yolo_weights_path)" />
>     <param name="config_path"                               value="$(arg yolo_config_path)" />
>   </node>
>   <node  ns="front_looking_camera" pkg="darknet_ros" type="darknet_ros" name="darknet_ros" output="screen" launch-prefix="$(arg launch_prefix)">
>     <param name="weights_path"                              value="$(arg yolo_weights_path)" />
>     <param name="config_path"                               value="$(arg yolo_config_path)" />
>   </node>
> 
>  
> </launch>
> ```

Is the directory structure as follows :  /front_looking_camera/darknet_ros/darknet_ros