leggedrobotics/darknet_ros #142: always waiting for image

rcpilotp51, 2019-02-10:
I have yolo running on a server and the robot publishing a usb camera topic `/usb_cam/image_raw` - i am able to see the image in rviz, but I get a continuous "waiting for image" message from the darknet_ros package after i configure the ros.yml file to subscribe to the same topic. I tried running all the launch files with no success.
Im not using a GPU, yet - I was just testing with a CPU but it cant seem to subscribe to the image topic.

Thanks for your help.

saipran7777, 2019-02-19:
I am facing the same issue. Did you find any fix to the problem?

albertogiacomini, 2019-03-21:
If you are running on multiple machines pay attention they are working on the same network

pedrobagatin, 2019-04-05:
Go to ros.yaml and change enable_opencv to false it will show detect objects % in the console.
Now I am trying to figure it out how to make it to display rectangle around detected images.

saipraneethd, 2019-04-08:
I can see the topic `/camera/image_raw` in `rostopic list` and I also checked `rostopic echo /camera/image_raw`. I also changed `ros.yaml` to subscribe to `/camera/image_raw`. The darknet_ros is still "Waiting for image". I am using Ubuntu bash on windows.

CDRE-LXIV, 2019-10-29:
I got around this by using cv_camera instead of darknet_ros's default node. I installed and ran cv_camera, then changed the camera topic in darknet_ros's ros.yaml to "topic: /cv_camera/image_raw". After that I launched darket_ros's yolo node and everything worked fine.

YangSiri, 2019-11-09:
Hi, we share the same issue. Should we change the codes below in src/YoloObjectDetector.cpp ?

 _nodeHandle_.param("subscribers/camera_reading/topic", cameraTopicName,
                    std::string("/camera/image_raw"));
  nodeHandle_.param("subscribers/camera_reading/queue_size", cameraQueueSize, 1);
  nodeHandle_.param("publishers/object_detector/topic", objectDetectorTopicName,
                    std::string("found_object"));
  nodeHandle_.param("publishers/object_detector/queue_size", objectDetectorQueueSize, 1);
  nodeHandle_.param("publishers/object_detector/latch", objectDetectorLatch, false);
  nodeHandle_.param("publishers/bounding_boxes/topic", boundingBoxesTopicName,
                    std::string("bounding_boxes"));
  nodeHandle_.param("publishers/bounding_boxes/queue_size", boundingBoxesQueueSize, 1);
  nodeHandle_.param("publishers/bounding_boxes/latch", boundingBoxesLatch, false);
  nodeHandle_.param("publishers/detection_image/topic", detectionImageTopicName,
                    std::string("detection_image"));
  nodeHandle_.param("publishers/detection_image/queue_size", detectionImageQueueSize, 1);
  nodeHandle_.param("publishers/detection_image/latch", detectionImageLatch, true);

  imageSubscriber_ = imageTransport_.subscribe(cameraTopicName, cameraQueueSize,
                                               &YoloObjectDetector::cameraCallback, this);_

And , hi, @CDRE-LXIV  could you explain more about the cv_camera?

Looking forward to your help. THANKS

CDRE-LXIV, 2019-11-10:
I installed the ROS package cv_camera (link: http://wiki.ros.org/cv_camera), which I run using "rosrun cv_camera cv_camera_node". That publishes several topics, the one we want is: "/cv_camera/image_raw".

I edited two files in darknet_ros:  
1.) In the launch folder, open darket_ros.launch: comment out the existing camera topic:   
<?ignore
<arg name="image" default="/camera/rgb/image_raw" />
?>
and insert:
<arg name="image" default="/cv_camera/image_raw" />.

2.) In the config folder, open ros.yaml: change the camera reading topic to: 
topic: /cv_camera/image_raw

Now if you launch darknet_ros using darket_ros.launch it should look for the image topic created by the cv_camera package instead. Good luck!

YangSiri, 2019-11-10:
THANKS @CDRE-LXIV 
I'm gonna try.

xiaofeiii, 2020-01-10:
Hello, I encountered the same problem, I use the csi camera on jetsonnano, but it keeps showing waiting for image

Gaurav37, 2020-11-26:
> I installed the ROS package cv_camera (link: http://wiki.ros.org/cv_camera), which I run using "rosrun cv_camera cv_camera_node". That publishes several topics, the one we want is: "/cv_camera/image_raw".
> 
> I edited two files in darknet_ros:
> 1.) In the launch folder, open darket_ros.launch: change the existing camera topic:
> arg name="image" default="/camera/rgb/image_raw" /
> to
> arg name="image" default="/cv_camera/image_raw" /
> 
> 2.) In the config folder, open ros.yaml: change the camera reading topic to:
> topic: /cv_camera/image_raw
> 
> Now if you launch darknet_ros using darket_ros.launch it should look for the image topic created by the cv_camera package instead. Good luck!

This works with usb_cam also. Running,  roslaunch usb_cam usb_cam-test.launch and then it tells  me image location to be /usb_cam/image_raw  and updated same in arg name in darknet_ros.launch and 
camera_reading:
    topic: /usb_cam/image_raw #/camera/rgb/image_raw in ros.yaml in darknet_ros/darknet_ros/config/ folder.
Then roslaunch darknet_ros darknet_ros.launch and works perfetctly. Its slow on cpu though!

AnukritiSinghh, 2021-07-23:
@rcpilotp51 hey I am very new to this. Can please help me how do you view images in rviz?
Also I tried updating the camera reading topic but I still face the same problem

Albert337, 2021-12-15:
> This works with usb_cam also. Running, roslaunch usb_cam usb_cam-test.launch and then it tells me image location to be /usb_cam/image_raw and updated same in arg name in darknet_ros.launch and [...]

I tried to use cv_camera or usb_cam bag, and i fixed the code as instructions, but the error still occured which is "waiting for image" and nothing shows.Is there any other ideas to slove it? thanks!

adhilcodes, 2022-10-11:
(1) How I solved the issue is, I installed the `usb_cam` module and updated my `ros.yaml` file as below
(2) Also update your `darknet_ros.launch file` with  **topic: /usb_cam/image_raw**

`subscribers:

  camera_reading:
    **topic: /usb_cam/image_raw**
    queue_size: 1

actions:

  camera_reading:
    name: /darknet_ros/check_for_objects

publishers:

  object_detector:
    topic: /darknet_ros/found_object
    queue_size: 1
    latch: false

  bounding_boxes:
    topic: /darknet_ros/bounding_boxes
    queue_size: 1
    latch: false

  detection_image:
    topic: /darknet_ros/detection_image
    queue_size: 1
    latch: true

image_view:

  enable_opencv: true
  wait_key_delay: 1
  enable_console_output: true`

ireemmozcan, 2022-10-26:
> I tried to use cv_camera or usb_cam bag, and i fixed the code as instructions, but the error still occured which is "waiting for image" and nothing shows.Is there any other ideas to slove it? thanks!

I have same issue. What should I do ? Thanks.

ireemmozcan, 2022-11-09:
> I have same issue. What should I do ? Thanks.

Hi. I solved my problem. I am using realsense D435i camera. I changed topic : **"/camera/color/image_raw/".** It worked