leggedrobotics/darknet_ros #144: error about darknet_ros.launch

QiuYilin, 2019-02-15:
I met this problem when I run darknet_ros.launch
[image]

I modified the makefie like this and maked it before I catkin_make
[image]

I don't know what's wrong.

Chrislzy1993, 2019-02-20:
maybe something wrong with cuda, did u install cuda correctly???

NPMN, 2019-05-16:
Reinstall either GPU driver and/or CUDA

anuj0707, 2019-12-27:
I am also getting the same error. What is the solution?

dippatel1994, 2021-05-02:
If it is solved by anyone, can you please share the resolution steps? stuck on same error.