leggedrobotics/darknet_ros #146: How to get BoundingBOX msgs in another node?

QiuYilin, 2019-02-22:
I try to  write a node to receive and show data of Bounding Box like below.
#include "ros/ros.h"
#include <darknet_ros_msgs/BoundingBoxes.h>
#include <darknet_ros_msgs/BoundingBox.h>
using namespace std;
void msgCallback(const darknet_ros_msgs::BoundingBoxes::ConstPtr& msg)

{
    cout<<"Bouding Boxes (header):" << msg->header <<endl;
	cout<<"Bouding Boxes (image_header):" << msg->image_header <<endl;
	cout<<"Bouding Boxes (Class):" << msg->bounding_boxes[0].Class <<endl;
	cout<<"Bouding Boxes (xmin):" << msg->bounding_boxes[0].xmin <<endl;
	cout<<"Bouding Boxes (xmax):" << msg->bounding_boxes[0].xmax <<endl;
	cout<<"Bouding Boxes (ymin):" << msg->bounding_boxes[0].ymin <<endl;
	cout<<"Bouding Boxes (ymax):" << msg->bounding_boxes[0].ymax <<endl;
	cout << "\033[2J\033[1;1H";     // clear terminal
}
    int main(int argc, char **argv)
    {
        ros::init(argc,argv,"coord_pixel_subscriber");
        ros::NodeHandle nh;
        ros::Subscriber ros_coord_pixel_sub = nh.subscribe("/darknet_ros/bounding_boxes",100,msgCallback);
        ros::spin();
        return 0;
    }

I have modify the cmakelist.txt and package.xml

cmakelist.txt
cmake_minimum_required(VERSION 2.8.3)
project(coord_tran)

find_package(catkin REQUIRED COMPONENTS
  geometry_msgs
  roscpp
  rospy
  std_msgs
  darknet_ros
  message_generation
)

catkin_package(
  INCLUDE_DIRS
  include
  LIBRARIES coord_tran
  CATKIN_DEPENDS geometry_msgs roscpp rospy std_msgs message_runtime 
  #DEPENDS system_lib
)

include_directories(
  include ${catkin_INCLUDE_DIRS}
  #include
  ${catkin_INCLUDE_DIRS}
)

add_executable(coord_tran src/coord_pixel_subscriber.cpp)
add_dependencies(coord_tran ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
target_link_libraries(coord_tran ${catkin_LIBRARIES})

package.xml:
<?xml version="1.0"?>
<package format="2">
  <name>coord_tran</name>
  <version>0.0.0</version>
  <description>The coord_tran package</description>

  
  <maintainer email="qiuyilin@todo.todo">qiuyilin</maintainer>

  <license>TODO</license>

  
  <build_depend>darknet_ros</build_depend>
  <buildtool_depend>catkin</buildtool_depend>
  <build_depend>geometry_msgs</build_depend>
  <build_depend>roscpp</build_depend>
  <build_depend>rospy</build_depend>
  <build_depend>std_msgs</build_depend>
  <build_export_depend>geometry_msgs</build_export_depend>
  <build_export_depend>roscpp</build_export_depend>
  <build_export_depend>rospy</build_export_depend>
  <build_export_depend>std_msgs</build_export_depend>
  <build_depend>message_generation</build_depend>
  <exec_depend>geometry_msgs</exec_depend>
  <exec_depend>roscpp</exec_depend>
  <exec_depend>rospy</exec_depend>
  <exec_depend>std_msgs</exec_depend>
  <exec_depend>darknet_ros</exec_depend>
  <exec_depend>message_generation</exec_depend>
  <exec_depend>message_runtime</exec_depend>
  
  <export>
    

  </export>
</package>

but when I run catkin_make, these mistakes happen
[100%] Built target darknet_ros
/usr/bin/ld: cannot find -lcudart
/usr/bin/ld: cannot find -lcublas
/usr/bin/ld: cannot find -lcurand
collect2: error: ld returned 1 exit status
coord_tran/CMakeFiles/coord_tran.dir/build.make:216: recipe for target '/home/qiuyilin/catkin_ws/devel/lib/coord_tran/coord_tran' failed
make[2]: *** [/home/qiuyilin/catkin_ws/devel/lib/coord_tran/coord_tran] Error 1
CMakeFiles/Makefile2:9146: recipe for target 'coord_tran/CMakeFiles/coord_tran.dir/all' failed
make[1]: *** [coord_tran/CMakeFiles/coord_tran.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

I don't know how to solve it.

Comedian1926, 2019-03-18:
我遇到了同样的问题，请问你解决了吗:)

Comedian1926, 2019-03-18:
cmakelist.txt
cmake_minimum_required(VERSION 2.8.3)
project(coord_tran)

find_package(catkin REQUIRED COMPONENTS
geometry_msgs
roscpp
rospy
std_msgs
darknet_ros
message_generation
)

中删除darknet_ros，添加darknet_ros_msgs试试

Shame-fight, 2019-10-28:
> cmakelist.txt
> cmake_minimum_required(VERSION 2.8.3)
> project(coord_tran)
> 
> find_package(catkin REQUIRED COMPONENTS
> geometry_msgs
> roscpp
> rospy
> std_msgs
> darknet_ros
> message_generation
> )
> 
> 中删除darknet_ros，添加darknet_ros_msgs试试

I also want to get BoundingBOX msgs in another node,do you have finished it? can you give me some advices.