leggedrobotics/darknet_ros #147: libdarknet_ros_lib.so:  undefined reference to *

koalamaster, 2019-03-06:
[ 93%] Linking CXX executable /home/nvidia/BLland_ws/devel/lib/darknet_ros/darknet_ros
[ 98%] Built target AirLib
[ 98%] Built target capture_from_ue
/home/nvidia/BLland_ws/devel/lib/libdarknet_ros_lib.so: undefined reference to `open_video_stream'
/home/nvidia/BLland_ws/devel/lib/libdarknet_ros_lib.so: undefined reference to `show_image_cv'
/home/nvidia/BLland_ws/devel/lib/libdarknet_ros_lib.so: undefined reference to `run_isegmenter'
/home/nvidia/BLland_ws/devel/lib/libdarknet_ros_lib.so: undefined reference to `load_image_cv'
/home/nvidia/BLland_ws/devel/lib/libdarknet_ros_lib.so: undefined reference to `make_iseg_layer'
/home/nvidia/BLland_ws/devel/lib/libdarknet_ros_lib.so: undefined reference to `ipl_into_image'
/home/nvidia/BLland_ws/devel/lib/libdarknet_ros_lib.so: undefined reference to `make_window'
/home/nvidia/BLland_ws/devel/lib/libdarknet_ros_lib.so: undefined reference to `ipl_to_image'
/home/nvidia/BLland_ws/devel/lib/libdarknet_ros_lib.so: undefined reference to `get_image_from_stream'

koalamaster, 2019-03-06:
where is the function "ipl_into_image" ?

koalamaster, 2019-03-06:
solved

closed 2019-03-06 by koalamaster

FiveHamburgers, 2019-09-23:
Hi, i have the same problem with you, could you tell me how do you solve the 'undefined reference to `ipl_into_image'? Thanks a lot!

dengjianbo3, 2019-12-06:
Hey, could you tell me how to solve this problem, thanks

Miguel-GitHub, 2019-12-18:
I have exactly the same problem. Please, if anyone knows the solution, it would be appreciated if you published it. Thank you.