leggedrobotics/darknet_ros #148: how to detect multiple class in BoundingBox msgs

nutywna, 2019-03-08:
I tried to  use this code for separate BoundingBox msgs but i have problem to detect multiple class 
I use this code
`#include "ros/ros.h"
#include <std_msgs/Int8.h>
#include <darknet_ros_msgs/BoundingBoxes.h>
#include <darknet_ros_msgs/BoundingBox.h>
using namespace std;
int numberofclass=0;
void msgCallback(const darknet_ros_msgs::BoundingBoxes::ConstPtr& msg)
{
  cout << "------------------------------" << endl;
for(int i=0 ; i < sizeof(msg); i++){
cout<<"Bouding Boxes (Class):" << msg->bounding_boxes[i].Class <<endl;
}
}

int main(int argc, char **argv)
{
ros::init(argc,argv,"coord_pixel_subscriber");
ros::NodeHandle nh;

ros::Subscriber ros_coord_pixel_sub = nh.subscribe("/darknet_ros/bounding_boxes",100,msgCallback);

ros::spin();
return 0;
}`
and this is output
------------------------------
Bouding Boxes (Class):turnleftsign
Segmentation fault (core dumped)

I don't know why is appear Segmentation fault (core dumped)
please help me thank you very much

shengje, 2019-06-04:
You need to use `msg->bounding_boxes.size()` instead of `sizeof(msg)` in the for loop
`for(int i=0 ; i < msg->bounding_boxes.size(); i++){
cout<<"Bouding Boxes (Class):" << msg->bounding_boxes[i].Class <<endl;
}`

adamkrawczyk, 2019-07-11:
void darknetCallback(const darknet_ros_msgs::BoundingBoxes &bb_msg) {
  int size = bb_msg.bounding_boxes.size(); // ros msgs are mapped onto std::vector
}