leggedrobotics/darknet_ros #152: Unable to use own weight

ChunJyeBehBeh, 2019-04-12:
Hello,

I have use [this model](https://github.com/eriklindernoren/PyTorch-YOLOv3) to train my own dataset. I have follow the steps under "Use you own detection objects" but seems like don't work. Is it necessary to get the weights using C++ model to train?

Besides, when I run the code, the terminal show the below
```
Cannot load image "/home/eee/TIM/src/darknet_ros/darknet/data/labels/86_0.png"
Cannot load image "/home/eee/TIM/src/darknet_ros/darknet/data/labels/87_0.png"
Cannot load image "/home/eee/TIM/src/darknet_ros/darknet/data/labels/88_0.png"
Cannot load image "/home/eee/TIM/src/darknet_ros/darknet/data/labels/89_0.png"
Cannot load image "/home/eee/TIM/src/darknet_ros/darknet/data/labels/90_0.png"
Cannot load image "/home/eee/TIM/src/darknet_ros/darknet/data/labels/91_0.png"
Cannot load image "/home/eee/TIM/src/darknet_ros/darknet/data/labels/92_0.png"
Cannot load image "/home/eee/TIM/src/darknet_ros/darknet/data/labels/93_0.png"
Cannot load image "/home/eee/TIM/src/darknet_ros/darknet/data/labels/94_0.png"
Cannot load image "/home/eee/TIM/src/darknet_ros/darknet/data/labels/95_0.png"
```

How can I disable this? Thx

He-Rong, 2019-05-14:
Did you solve this problem ? Or any idea of this problem?
I met the same problem after the image pop out.
 Thank you!

Martinpanzy, 2019-05-16:
Same Issue

XueMingXiao, 2020-05-10:
> Hello,
> 
> I am successfully load the weight that I train for my own object.
> 
> ```
> Loading weights from /home/beh/TIM/src/darknet_ros/darknet_ros/darknet_ros/yolo_network_config/weights/test.weights...Done!
> Waiting for image.
> [darknet_ros-1] process has died [pid 9285, exit code -11, cmd /home/beh/TIM/devel/lib/darknet_ros/darknet_ros __name:=darknet_ros __log:=/home/beh/.ros/log/eb20dd32-5ff3-11e9-89a1-80a589799ef1/darknet_ros-1.log].
> log file: /home/beh/.ros/log/eb20dd32-5ff3-11e9-89a1-80a589799ef1/darknet_ros-1*.log
> all processes on machine have died, roslaunch will exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> ```
> 
> After the image pop out, it immediately close and report the error above. May I know how to solve this problem?

I met this issue too. I solved it by changing to darknet_ros.launch instead of yolo_v3.launch. That maybe help