leggedrobotics/darknet_ros #158: yolov3.yaml if classes removed, lower performance

chbloca, 2019-05-08:
Since I would like to use this package just for detecting certain object categories. I just delete from the list in yolov3.yaml the objects I do not want to appear. However, I have notice this does not work as it should since the YOLO will anyways make detections on such objects but will label them with the survivals on the list.

In addition, the quality of detection of the surviving objects in the list decreases unexplicably.

For instance, if I just want to detect persons, there will be a lot of Positive Negatives.
[image]

rekalantar, 2019-06-20:
I think you meant false positives?

chbloca, 2019-06-22:
> 
> 
> I think you meant false positives?

No, True Negatives, it is different stuff. See the picture above.

fiorano10, 2019-07-15:
@chbloca You can't just delete labels...you need to retrain the model for only those objects, look at @AlexeyAB repo.