leggedrobotics/darknet_ros #162: Boundingbox synchronization problem with action call

TobiMiller, 2019-05-28:
Hey,

i send the image from the image callback as an action goal and wait for the result. i draw the bounding box with rectangle into the opencv image. the bounding box is shifted during movements. as if the bounding box belonged to an old image. 
can anyone confirm that?

i don't use the topic method, because i can't synchronize the box properly.

MayurT, 2019-06-04:
Hi Tobi

I picked up the same issue when trying to use the action goal recently, with the response containing detection results for an old image. I've been working on a fix - will push it soon, just need to refine the code. It seems to be working for me so far, but will let you know when it's pushed and perhaps you could do some tests on your end as well?

Mayur

TobiMiller, 2019-06-04:
@MayurT 
Wow! That would be awesome :) is it already in your fork? i would like to use it as soon as possible. i need it for my master thesis, which is a complete different topic.

MayurT, 2019-06-05:
It's in my fork now :) You can check it out at the ``fix/action_server_result`` branch. There's still some minor mods I need to do, but this should work for now. Please let me know if you have feedback from your tests.

By the way, you mentioned you don't use the topic method because you can't synchronize the box. Do you mean you are not able to match the bounding box results with the specific images they correspond to?

TobiMiller, 2019-06-05:
@MayurT 
correct.
i have a ros image topic which is processed in an opencv node. i want to use the boundingbox to get a rough estimation of the position of an object in my picture. but with subscriber/publisher i dont know how to handle the synchronization between the two callbacks. one callback is the image callback and the other would be the darknet_ros callback. 

i think for this task is a more synchronous behaviour better

MayurT, 2019-06-06:
You can use the timestamp in the ``image_header`` of the BoundingBoxes msg
for synchronization.The way we handle it on a project I'm working on is to
have the image callback store images on a buffer, then the bounding box
callback runs through the image buffer until it finds the image with a
matching stamp to the ``image_header`` stamp in the bounding box, and
process accordingly.

Of course, you could just stick to using the action goal method if you
don't need the faster throughput of topics.

TobiMiller, 2019-06-07:
@MayurT 

how do you buffer the topics?

MayurT, 2019-06-07:
@TobiMiller 
You can use a data structure like a vector (we use a Boost circular buffer to limit memory usage) and just copy the image into it from inside the image callback. Then when you get a bounding box, iterate through the buffer object until you find the image with a matching timestamp.

TobiMiller, 2019-06-11:
thank you very much. i will test a buffer system also.