leggedrobotics/darknet_ros #163: Why is the detection_image rate so much higher than the camera image rate ?

elpidiovaldez, 2019-06-05:
I am having trouble understanding the rates reported by 'rostopic hz':

```
rostopic hz /raspicam_node/image/compressed
subscribed to [/raspicam_node/image/compressed]
average rate: 0.622
	min: 1.608s max: 1.608s std dev: 0.00000s window: 2

rostopic hz /raspicam_node/image
subscribed to [/raspicam_node/image]
average rate: 4.635
	min: 0.200s max: 0.235s std dev: 0.01266s window: 5

rostopic hz /darknet_ros/detection_image
subscribed to [/darknet_ros/detection_image]
average rate: 43.037
	min: 0.013s max: 0.030s std dev: 0.00303s window: 43

rostopic hz /darknet_ros/bounding_boxes
subscribed to [/darknet_ros/bounding_boxes]
average rate: 42.849
	min: 0.020s max: 0.030s std dev: 0.00164s window: 42
```
So the compressed image messages are under 1 hz.  I imagine a few frames are bundled into one message and sent together.  When they are decompressed the image rate is nearly 5hz.  That makes sense, but why is the detection image rate so much higher than the image rate ?  Is darknet_ros pointlessly running the detector nearly 10 times on the same image ?  The bounding boxes are arriving at about the same rate as the detection images, and inspecting them shows about 10 repetitions of the same bounding box data.  Isn't this wasting a lot of GPU work and watts ?  Is there anyway to ask darknet_ros to run once per video frame ?

My equipment is Raspberry Pi 3b + Pi cam V2 sending compressed images to a desktop with a GTX 1080 ti running the image decompression and darknet node.

elpidiovaldez, 2019-07-07:
There are a few of posts concerning this behaviour. I have looked at the code but I do not understand it well. Nevertheless I have managed to implement a 'fix' that appears to work.

I have a 1080ti GPU and I found that identical bounding boxes were being published around 5 times for each video frame (camera running at about 10fps). GPU usage was about 87%. After implementing the fix bounding boxes are published at the frame rate from distinct frames. GPU usage dropped to 26% or less. GPU Memory usage unchanged.

I am sure that a much more elegant solution could be produced by somebody who actually understood the code. I have never published code on Github and am not sure what I need to do to release the fix for inspection by a wider audience. If anybody is interested I can send them the code.

leoll2, 2019-07-15:
> There are a few of posts concerning this behaviour. I have looked at the code but I do not understand it well. Nevertheless I have managed to implement a 'fix' that appears to work.
>
> I have a 1080ti [...]

If you want to submit your fix, just fork this project, apply your changes and create a pull request.
Otherwise, you may simply discuss in this issue the modifications you've made, provided they're simple enough.

fiorano10, 2019-07-15:
@elpidiovaldez Can you at least fork the project and implement the fix, even if you don't want to create a PR?

elpidiovaldez, 2019-07-16:
@fiorano10 @leoll2 This discussion seems to have moved to issue #150.  I have given my changes in that thread, and there is another fix which has been proposed there too (but it did not work for me).

fiorano10, 2019-07-16:
@elpidiovaldez I saw that too, I’ll implement your fix and update you.

elpidiovaldez, 2019-07-16:
@fiorano10 cool !  It might be worth trying the other fix too.  It seems to be simpler to implement since you only need to check out 1 file.  It did not work for me, but I have a bad case - a slow camera with a storming gpu.

fiorano10, 2019-07-16:
@elpidiovaldez I have made a PR for your fix, I'll test both fixes tomorrow, and update you. Thanks!