leggedrobotics/darknet_ros #166: Error building on Ubuntu 18 with CUDA

tleyden, 2019-06-23:
## Environment

* Ubuntu 18
* GEforce GTX 1070
* Nvidia CUDA drivers installed

## Repro steps

* Git clone repo (recursively)
* `catkin_make`

## Errors

```
[ 46%] Built target darknet_ros_msgs_generate_messages

In file included from /usr/local/cuda/include/host_config.h:50:0,
                 from /usr/local/cuda/include/cuda_runtime.h:78,
                 from <command-line>:0:
/usr/local/cuda/include/crt/host_config.h:119:2: error: #error -- unsupported GNU version! gcc versions later than 6 are not supported!
 #error -- unsupported GNU version! gcc versions later than 6 are not supported!

```

## Workaround

I commented the CUDA related steps in `darknet_ros/darknet_ros/CMakeLists.txt` and it was able to build successfully.  

It would be nice to be able build this with CUDA support though.  Is my CUDA version out of date?

ericjohnson97, 2019-06-26:
Just ran into this issue. I think ubuntu 18.04 now comes with a GNU version > 6, a version CUDA 9 doesn't recognize. I was able to build the package by running 

```
catkin build -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER=/usr/bin/gcc-6
```
Hope this helps!

mipsan, 2019-06-26:
Ubuntu : 18.04 .02 (4.9.xxx)
cuda: 10.0 
GPU: Pascal gen (Jetson TX2) 
gcc: 7.3.0 

I built and tested this today.

rolker, 2019-10-16:
I had the same error with the version of cuda provided by the package manager. That problem went away when I installed the latest cuda from NVidia.

I still don't get a clean build. I am now having issues with OpenCV which I didn't get when building without cuda.

BhooshanDeshpande, 2020-06-19:
Hello All, 

i have the same issue here, 
`#error -- unsupported GNU version! gcc versions later than 6 are not supported!`

My specs are - 
Ubuntu 18.04 
GPU - GeForce GTX 1660Ti 
gcc: 7.5.0
CUDA 10.2 

I believe every Ubuntu 18.04 has a gcc version of 7 and hence the package has issues building with  CUDA support.  My entire error is - 

`In file included from /usr/include/host_config.h:50:0,
                 from /usr/include/cuda_runtime.h:78,
                 from <command-line>:0:
/usr/include/crt/host_config.h:121:2: error: #error -- unsupported GNU version! gcc versions later than 6 are not supported!
 #error -- unsupported GNU version! gcc versions later than 6 are not supported!
  ^~~~~
In file included from /usr/include/host_config.h:50:0,
                 from /usr/include/cuda_runtime.h:78,
                 from <command-line>:0:
/usr/include/crt/host_config.h:121:2: error: #error -- unsupported GNU version! gcc versions later than 6 are not supported!
 #error -- unsupported GNU version! gcc versions later than 6 are not supported!
  ^~~~~
`
Let me know if there is any work-around. I am going to try what @ericjohnson97 said, will post the results soon.

ican9595, 2021-05-08:
Hello All, 

any updates on that Issue? I am having the same problem.
I tried the same as @BhooshanDeshpande but when I run the darknet_ros.launch file now I get following error:
ERROR: cannot launch node of type [darknet_ros/darknet_ros]: Cannot locate node of type [darknet_ros] in package [darknet_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)

asibarr2, 2022-03-31:
> Just ran into this issue. I think ubuntu 18.04 now comes with a GNU version > 6, a version CUDA 9 doesn't recognize. I was able to build the package by running
> 
> ```
> catkin build -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_COMPILER=/usr/bin/gcc-6
> ```
> 
> Hope this helps!

Worked for me! Thanks!