leggedrobotics/darknet_ros #167: detection window to roshost side

mipsan, 2019-06-26:
Dear,

I built and tested darknet_ros today on NVIDIA Jetson TX2.
It runs pretty well much more than I expected.

Then I want to display this detection window on roshost side, if possible.
And darknet running console might be created and displayed another terminal.
I am not able to monitor ROS logs. 

Thanks,

mipsan, 2019-06-26:
It works!

I launch darknet_ros in other terminal, 
`$ roslaunch darknet_ros darknet_ros.launch `

This clears two issues at a time!

Thanks,

alexis-pepe, 2020-08-15:
Hello, I am also using TX2 to run Darknet, using command roslaunch darknet_ros darknet_ros.Launch, using Yolov3, , FPS is only about 0.4, when I use Yolov2-tiny, FPS is raised to about 4, but the effect is still not ideal, I also modified in Makefile, GPU=1, CUDNN=1, OPENCV=1, I used JetPack3.0, CUDA8,
It looks like Cudnn isn't working. Can we talk about it?

akshaynarisetti, 2021-08-22:
@alexis-pepe  I have the same issue the 0.4 fps, Let me know If you found a fix