leggedrobotics/darknet_ros #172: 0 CUDA Error: unknown error

1329844541, 2019-07-22:
0 CUDA Error: unknown error
darknet_ros: /home/mou/catkin_ws1/src/darknet_ros/darknet/src/cuda.c:36: check_error: Assertion `0' failed.
I met a problem like this,do you know how to handle it? Thank you.

1329844541, 2019-07-22:
process[darknet_ros-2]: started with pid [16297]
[ INFO] [1563792551.715212499]: [YoloObjectDetector] Node started.
[ INFO] [1563792551.718412362]: [YoloObjectDetector] Xserver is running.
[ INFO] [1563792551.719387842]: [YoloObjectDetector] init().
YOLO V3
layer     filters    size              input                output
    0 CUDA Error: unknown error
darknet_ros: /home/mou/catkin_ws1/src/darknet_ros/darknet/src/cuda.c:36: check_error: Assertion `0' failed.
[darknet_ros-2] process has died [pid 16297, exit code -6, cmd /home/mou/catkin_ws1/devel/lib/darknet_ros/darknet_ros __name:=darknet_ros __log:=/home/mou/.ros/log/567163c4-ac6e-11e9-8fac-9c305b37cb2b/darknet_ros-2.log].
log file: /home/mou/.ros/log/567163c4-ac6e-11e9-8fac-9c305b37cb2b/darknet_ros-2*.log

This is all massage.

fiorano10, 2019-10-28:
Can you copy & paste `nvcc -V` here?

renatojmsdh, 2020-03-31:
Did anyone find a solution? 

I'm having the same issue my PC with CUDA 10.0. Neither the weights or configuration are loaded.

```
[ INFO] [1585614661.644136477]: [YoloObjectDetector] Node started.
[ INFO] [1585614661.680486156]: [YoloObjectDetector] Xserver is running.
[ INFO] [1585614661.681623332]: [YoloObjectDetector] init().
YOLO V3
layer     filters    size              input                output
    0 CUDA Error: unknown error
CUDA Error: unknown error: Success
```
@fiorano10 my **nvcc -V:**

```
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2018 NVIDIA Corporation
Built on Sat_Aug_25_21:08:01_CDT_2018
Cuda compilation tools, release 10.0, V10.0.130

```

closed 2021-06-08 by tomlankhorst