leggedrobotics/darknet_ros #175: CUDA Error: invalid argument   CUDA Error: invalid argument: File exists

Magical-E, 2019-08-06:
I have already install cuda and cudnn,but when launch the file it display:

>  mask_scale: Using default '1.000000'
Loading weights from /home/magical/astra_ws/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolov2-tiny.weights...**CUDA Error: invalid argument
CUDA Error: invalid argument: File exists**
[darknet_ros-1] process has died [pid 8073, exit code 255, cmd /home/magical/astra_ws/devel/lib/darknet_ros/darknet_ros __name:=darknet_ros __log:=/home/magical/.ros/log/837e1408-b816-11e9-8c85-505bc2deab21/darknet_ros-1.log].
log file: /home/magical/.ros/log/837e1408-b816-11e9-8c85-505bc2deab21/darknet_ros-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Here is my cuda and cudnn:
> magical@magical-Lenovo-Legion-Y7000:~ $ nvidia-smi 
Tue Aug  6 15:07:29 2019       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 430.40       Driver Version: 430.40       CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 105...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   46C    P0    N/A /  N/A |    256MiB /  4040MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+                                                                      
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1218      G   /usr/lib/xorg/Xorg                           135MiB |
|    0      1749      G   /usr/bin/gnome-shell                         117MiB |
+-----------------------------------------------------------------------------+

> magical@magical-Lenovo-Legion-Y7000:~ $ cat /usr/local/cuda/include/cudnn.h | grep CUDNN_MAJOR -A 2
#define CUDNN_MAJOR 7
#define CUDNN_MINOR 6
#define CUDNN_PATCHLEVEL 2
--
#define CUDNN_VERSION (CUDNN_MAJOR * 1000 + CUDNN_MINOR * 100 + CUDNN_PATCHLEVEL)
#include "driver_types.h"

I would appreciate it if anyone could help me.

AndLydakis, 2020-02-28:
Just wondering, did you ever solve this and happen to remember the solution? Cheers!

Magical-E, 2020-02-29:
> Just wondering, did you ever solve this and happen to remember the solution? Cheers!

Instead of git clone,I chose to download the zip,then problems solved.

Fenix0817, 2020-03-05:
Hi,

I have the same problem.

Loading weights from /home/aegh/catkin_ws/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolov3110000.weights...Done!
Waiting for image.
**CUDA Error: invalid argument
CUDA Error: invalid argument: Success
QObject::~QObject: Timers cannot be stopped from another thread**
[kitti_darknet_ros-1] process has died [pid 20018, exit code -11, cmd /home/aegh/catkin_ws/devel/lib/darknet_ros/darknet_ros __name:=darknet_ros __log:=/home/aegh/.ros/log/1ea4e58a-5ec3-11ea-923e-54bf64736bb8/darknet_ros-1.log].
log file: /home/aegh/.ros/log/1ea4e58a-5ec3-11ea-923e-54bf64736bb8/darknet_ros-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

Best regards

Fenix0817, 2020-03-09:
Hello everybody,

I did find the problem. I only updated the driver of my graphic card and the Cuda version. This is my machine setup:

Ubuntu 16.04
GeForce GTX 1080
Driver 440.33
Cuda 10.2

Best regards.