leggedrobotics/darknet_ros #177: Port for ROS2?

NovoG93, 2019-08-27:
Will there be a port for ROS2?

cardboardcode, 2019-11-24:
Hi @NovoG93 , I am interested in this issue as well. 

@mbjelonic , are there any plans for this port? If not, I am willing to invest some bandwidth to address this issue.

fmrico, 2020-01-14:
Hi @NovoG93 @cardboardcode 

I have just ported darknet_ros to ROS2. To make a PR, @mbjelonic , could you make a ros2 (or the name you prefer) branch?

[image]

Enjoy!!!

mbjelonic, 2020-01-15:
Hi all,
thank you for porting the software to ROS2. I just created a new branch called ros2. Looking forward to the PR.
Best,
Marko

cardboardcode, 2020-01-15:
@fmrico 

Awesome work. Thank you so much. 

Will be checking it out soon.

fmrico, 2020-01-15:
Thanks!!!!

I have just made the PR in #206 

Best
Francisco

tomlankhorst, 2021-06-08:
See https://github.com/leggedrobotics/darknet_ros/tree/foxy

closed 2021-06-08 by tomlankhorst