leggedrobotics/darknet_ros #188: catkin_make  cuda error

YangSiri, 2019-10-09:
Hi, Contributors,  thanks for your codes. But i got a compile error. After i clone the codes and run catkin_make -DCMAKE_BUILD_TYPE=Release, some issue concerned about cuda occurred. My environment is cuda9.0. 

Error:
_-- Configuring done
CMake Error at /usr/local/share/cmake-3.12/Modules/FindCUDA.cmake:1816 (add_library):
  Cannot find source file:

    DARKNET_PATH-NOTFOUND/src/activation_layer.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  darknet_ros/darknet_ros/CMakeLists.txt:97 (cuda_add_library)

CMake Error at /usr/local/share/cmake-3.12/Modules/FindCUDA.cmake:1816 (add_library):
  No SOURCES given to target: darknet_ros_lib
Call Stack (most recent call first):
  darknet_ros/darknet_ros/CMakeLists.txt:97 (cuda_add_library)_

Looking forward to your help~

vincentzhouwenqi, 2019-10-11:
hi,i meet this problem too. Do you have solution now?

YangSiri, 2019-10-11:
not yet. waiting patiently...

fiorano10, 2019-10-28:
Did you do `git clone --recursive` when cloning this repo? It seems like you don't have darknet submodule cloned within darknet_ros

TheKursta, 2019-11-07:
Hi, same error. CUDA 10.0 and cuDNN 7.5 are correctly set up on my machine. Where  may I find a solution?

TheKursta, 2019-11-07:
> Hi, same error. CUDA 10.0 and cuDNN 7.5 are correctly set up on my machine. Where may I find a solution?

Hi, again. I was able to solve issue by purging incorrectly cloned repo and cloning it correctly using in manual specified SSH method.

Sorry for disturbance. :)

YangSiri, 2019-11-08:
Yes, @fiorano10 , thank you!

closed 2019-11-08 by YangSiri