leggedrobotics/darknet_ros #190: How to subscribe to the boundingbox topic and a camera input

jyp0802, 2019-10-16:
I'm trying to subscribe to the BoundingBoxes topic and a camera input from a node in a different package.

In the main function this is my code:
```
class MyClass
{
public:
  void MyCallback(const sensor_msgs::ImageConstPtr& mImage,const darknet_ros_msgs::BoundingBoxes& mBoxes);
};

int main(int argc, char **argv)
{
...
  message_filters::Subscriber<sensor_msgs::Image> sub_image(nh, "/camera/color/image_raw", 1);
  message_filters::Subscriber<darknet_ros_msgs::BoundingBoxes> sub_boxes(nh, "/darknet_ros/bounding_boxes", 1);
  typedef message_filters::sync_policies::ApproximateTime<sensor_msgs::Image, darknet_ros_msgs::BoundingBoxes> sync_pol;
  message_filters::Synchronizer<sync_pol> sync(sync_pol(10), sub_image,sub_boxes);
  sync.registerCallback(boost::bind(&MyClass::MyCallback,_1,_2));
...
}
```

But when I catkin_make this, I get an error which I think is because of the boost::bind on the **darknet_ros_msgs**.

```
/usr/include/boost/bind/bind.hpp:69:37: error: ‘void (MyClass::*)(const boost::shared_ptr<const sensor_msgs::Image_<std::allocator<void> > >&, const darknet_ros_msgs::BoundingBoxes_<std::allocator<void> >&)’ is not a class, struct, or union type
     typedef typename F::result_type type;

AND

/usr/include/boost/function/function_template.hpp:159:11: error: no match for call to ‘(boost::_bi::bind_t<boost::_bi::unspecified, boost::_bi::bind_t<boost::_bi::unspecified, void (MyClass::*)(const boost::shared_ptr<const sensor_msgs::Image_<std::allocator<void> > >&, const darknet_ros_msgs::BoundingBoxes_<std::allocator<void> >&), boost::_bi::list2<boost::arg<1>, boost::arg<2> > >, boost::_bi::list9<boost::arg<1>, boost::arg<2>, boost::arg<3>, boost::arg<4>, boost::arg<5>, boost::arg<6>, boost::arg<7>, boost::arg<8>, boost::arg<9> > >) (const boost::shared_ptr<const sensor_msgs::Image_<std::allocator<void> > >&, const boost::shared_ptr<const darknet_ros_msgs::BoundingBoxes_<std::allocator<void> > >&, const boost::shared_ptr<const message_filters::NullType>&, const boost::shared_ptr<const message_filters::NullType>&, const boost::shared_ptr<const message_filters::NullType>&, const boost::shared_ptr<const message_filters::NullType>&, const boost::shared_ptr<const message_filters::NullType>&, const boost::shared_ptr<const message_filters::NullType>&, const boost::shared_ptr<const message_filters::NullType>&)’
           BOOST_FUNCTION_RETURN((*f)(BOOST_FUNCTION_ARGS));
```

Basically, I'm not sure how I can subscribe to the BoundingBoxes topic.
If I change the BoundingBoxes related stuff to another "normal" topic like depth image, no errors occur.

Thanks in advance!

jyp0802, 2019-10-17:
Nevermind! 
I realized that I needed to add ```::ConstPtr``` to the input argument

```
void MyCallback(const sensor_msgs::ImageConstPtr& mImage,const darknet_ros_msgs::BoundingBoxes::ConstPtr& mBoxes);
```

closed 2019-10-17 by jyp0802

Shame-fight, 2020-04-16:
Can you share your code, I also want to subscribe to the BoundingBoxes topic and a camera input from a node. I am a newbie, if you can provide the code, it will be of great help to me. Thanks in advance.
@jyp0802

jyp0802, 2020-04-16:
`
/**
* This file is part of ORB-SLAM2.
*
* Copyright (C) 2014-2016 Raúl Mur-Artal <raulmur at unizar dot es> (University of Zaragoza)
* For more information see <https://github.com/raulmur/ORB_SLAM2>
*
* ORB-SLAM2 is free software: you can redistribute it and/or modify
* it under the terms of the GNU General Public License as published by
* the Free Software Foundation, either version 3 of the License, or
* (at your option) any later version.
*
* ORB-SLAM2 is distributed in the hope that it will be useful,
* but WITHOUT ANY WARRANTY; without even the implied warranty of
* MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
* GNU General Public License for more details.
*
* You should have received a copy of the GNU General Public License
* along with ORB-SLAM2. If not, see <http://www.gnu.org/licenses/>.
*/

#include<iostream>
#include<algorithm>
#include<fstream>
#include<chrono>

#include<ros/ros.h>
#include <ros/package.h>
#include <cv_bridge/cv_bridge.h>
#include <message_filters/subscriber.h>
#include <message_filters/time_synchronizer.h>
#include <message_filters/sync_policies/approximate_time.h>
#include "geometry_msgs/TransformStamped.h"
#include "tf/transform_datatypes.h"
#include <tf/transform_broadcaster.h>

#include<opencv2/core/core.hpp>

#include <darknet_ros_msgs/BoundingBoxes.h>

#include"System.h"

using namespace std;

class ImageGrabber
{
public:
    ImageGrabber(ORB_SLAM2::System* pSLAM):mpSLAM(pSLAM){}

    void GrabRGBD(const sensor_msgs::ImageConstPtr& msgRGB,const sensor_msgs::ImageConstPtr& msgD,const darknet_ros_msgs::BoundingBoxes::ConstPtr& msgObjects);

    ORB_SLAM2::System* mpSLAM;
};

int main(int argc, char **argv)
{
    ros::init(argc, argv, "rgbd_slam");
    ros::start();

    // Create SLAM system. It initializes all system threads and gets ready to process frames.
    ORB_SLAM2::System SLAM(argv[1],argv[2],ORB_SLAM2::System::RGBD,true,load_map,save_map);

    ImageGrabber igb(&SLAM);

    ros::NodeHandle nh;

    message_filters::Subscriber<sensor_msgs::Image> rgb_sub(nh, "/darknet_ros/original_image", 1);
    message_filters::Subscriber<sensor_msgs::Image> depth_sub(nh, "/camera/aligned_depth_to_color/image_raw", 1);
    message_filters::Subscriber<darknet_ros_msgs::BoundingBoxes> object_sub(nh, "/darknet_ros/bounding_boxes", 1);
    typedef message_filters::sync_policies::ApproximateTime<sensor_msgs::Image, sensor_msgs::Image, darknet_ros_msgs::BoundingBoxes> sync_pol;
    message_filters::Synchronizer<sync_pol> sync(sync_pol(10), rgb_sub,depth_sub,object_sub);
    sync.registerCallback(boost::bind(&ImageGrabber::GrabRGBD,&igb,_1,_2,_3));

    ros::spin();

    // Stop all threads
    SLAM.Shutdown();

    ros::shutdown();

    return 0;
}

void ImageGrabber::GrabRGBD(const sensor_msgs::ImageConstPtr& msgRGB,const sensor_msgs::ImageConstPtr& msgD,const darknet_ros_msgs::BoundingBoxes::ConstPtr& msgObjects)
{
    // Copy the ros image message to cv::Mat.
    cv_bridge::CvImageConstPtr cv_ptrRGB;
    try
    {
        cv_ptrRGB = cv_bridge::toCvShare(msgRGB);
    }
    catch (cv_bridge::Exception& e)
    {
        ROS_ERROR("cv_bridge exception: %s", e.what());
        return;
    }

    cv_bridge::CvImageConstPtr cv_ptrD;
    try
    {
        cv_ptrD = cv_bridge::toCvShare(msgD);
    }
    catch (cv_bridge::Exception& e)
    {
        ROS_ERROR("cv_bridge exception: %s", e.what());
        return;
    }

    int n = msgObjects->bounding_boxes.size();
    int x, y, w, h;
    std::vector<cv::Rect> objects;
    objects.reserve(n);
    for (int i = 0; i < n; i++)
    {
        x = msgObjects->bounding_boxes[i].xmin;
        y = msgObjects->bounding_boxes[i].ymin;
        w = msgObjects->bounding_boxes[i].xmax - x;
        h = msgObjects->bounding_boxes[i].ymax - y;
        objects.push_back(cv::Rect(x, y, w, h));
    }

    cv::Mat pose = mpSLAM->TrackRGBD(cv_ptrRGB->image,cv_ptrD->image,cv_ptrRGB->header.stamp.toSec(),objects);
}
`

This is the part where I got the camera inputs and the bounding boxes too. The code is from ORB_SLAM2 and I just modified it a bit to subscribe to the bounding boxes too. Hope it helps!

Shame-fight, 2020-04-16:
I am Glad to receive your reply. I want to subscribe to the bounding_boxes topic to use for the next step of the work. (Such as robot navigation). As described in this question:[https://github.com/leggedrobotics/darknet_ros/issues/146]
 sorry,I do n’t know how to modify it. As far as I know, I should modify several places?(liakecmakelist.txt package.xml ) Can you provide the revised code?thank you very much.

Shame-fight, 2020-04-16:
I am Glad to receive your reply. I want to subscribe to the bounding_boxes topic to use for the next step of the work. (Such as robot navigation). As described in this question:[https://github.com/leggedrobotics/darknet_ros/issues/146]
 sorry,I do n’t know how to modify it. As far as I know, I should modify several places?(liakecmakelist.txt package.xml ) Can you provide the revised code?thank you very much. @jyp0802