leggedrobotics/darknet_ros #192: How to test an image with zed camera?

kctoayo88, 2019-10-23:
Hello,
The default detecting way is keeping detect a real-time video from camera.
So I want to captured an image, then input it to YOLO model to detect objects with c++ node. 

Is there anyone know how to do it?
If you could provide me some example code, it will be nice to me!

Thanks!

YangSiri, 2019-11-09:
Hi, @kctoayo88 
I'm also testing using ZED.  I recorded the image data in rosbags. But i dont know how to detect it using this. I changed the topic name. It didnt work. Do you have any comments? 
Have tested this in real-time camera ?

Looking forward to your reply. THANKS

mtbsteve, 2019-12-15:
You need to have the ZED ROS wrapper installed and then launch it with 
`roslaunch zed_wrapper zed.launch`
in the darknet_ros.launch file set the desired zed node as input:
`<arg name="image" default="/zed/zed_node/left/image_rect_color" />`
Works perfectly fine for me.

closed 2021-06-08 by tomlankhorst