leggedrobotics/darknet_ros #193: cmake error

Levi568, 2019-10-29:
### I have CUDA 10.1 installed and tested it but I am having the following error while compiling it.
-------------------------------------------------------------------------------------------------------------------------
CMake Error at darknet_ros_lib_generated_maxpool_layer_kernels.cu.o.cmake:207 (message):
  Error generating
  /home/wei/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_maxpool_layer_kernels.cu.o

darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:126: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_maxpool_layer_kernels.cu.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_maxpool_layer_kernels.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 81%] Building NVCC (Device) object darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_im2col_kernels.cu.o
[ 82%] Built target ground_projection_generate_messages_lisp
[ 83%] Built target fleet_planning_generate_messages_py
[ 83%] Built target fleet_planning_generate_messages_nodejs
CMake Error at darknet_ros_lib_generated_activation_kernels.cu.o.cmake:207 (message):
  Error generating
  /home/wei/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_activation_kernels.cu.o

darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:63: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_activation_kernels.cu.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_activation_kernels.cu.o] Error 1
CMake Error at darknet_ros_lib_generated_crop_layer_kernels.cu.o.cmake:207 (message):
  Error generating
  /home/wei/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_crop_layer_kernels.cu.o

darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:70: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_crop_layer_kernels.cu.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_crop_layer_kernels.cu.o] Error 1
CMake Error at darknet_ros_lib_generated_avgpool_layer_kernels.cu.o.cmake:207 (message):
  Error generating
  /home/wei/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_avgpool_layer_kernels.cu.o

CMake Error at darknet_ros_lib_generated_deconvolutional_kernels.cu.o.cmake:207 (message):
  Error generating
  /home/wei/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_deconvolutional_kernels.cu.o

darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:77: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_avgpool_layer_kernels.cu.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_avgpool_layer_kernels.cu.o] Error 1
darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:84: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_deconvolutional_kernels.cu.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_deconvolutional_kernels.cu.o] Error 1
CMake Error at darknet_ros_lib_generated_blas_kernels.cu.o.cmake:207 (message):
  Error generating
  /home/wei/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_blas_kernels.cu.o

[ 83%] Built target ground_projection_generate_messages_py
darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:91: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_blas_kernels.cu.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_blas_kernels.cu.o] Error 1
CMake Error at darknet_ros_lib_generated_dropout_layer_kernels.cu.o.cmake:207 (message):
  Error generating
  /home/wei/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_dropout_layer_kernels.cu.o

darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:98: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_dropout_layer_kernels.cu.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_dropout_layer_kernels.cu.o] Error 1
[ 83%] Built target multivehicle_tracker_generate_messages_nodejs
[ 83%] Built target ground_projection_generate_messages_nodejs
[ 84%] Built target ground_projection_generate_messages_cpp
[ 84%] Built target ground_projection_generate_messages_eus
CMake Error at darknet_ros_lib_generated_col2im_kernels.cu.o.cmake:207 (message):
  Error generating
  /home/wei/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_col2im_kernels.cu.o

darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:105: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_col2im_kernels.cu.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_col2im_kernels.cu.o] Error 1
[ 85%] Built target multivehicle_tracker_generate_messages_py
CMake Error at darknet_ros_lib_generated_im2col_kernels.cu.o.cmake:207 (message):
  Error generating
  /home/wei/catkin_ws/build/darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/./darknet_ros_lib_generated_im2col_kernels.cu.o

darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:112: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_im2col_kernels.cu.o' failed
make[2]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/darknet_ros_lib_generated_im2col_kernels.cu.o] Error 1
CMakeFiles/Makefile2:10515: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 85%] Built target multivehicle_tracker_generate_messages_cpp
[ 85%] Built target multivehicle_tracker_generate_messages_lisp
[ 85%] Built target navigation_generate_messages_py
[ 86%] Built target multivehicle_tracker_generate_messages_eus
[ 86%] Built target navigation_generate_messages_eus
[ 86%] Built target navigation_generate_messages_nodejs
[ 86%] Built target navigation_generate_messages_lisp
[ 86%] Built target navigation_generate_messages_cpp
[ 86%] Built target object_detection_generate_messages_nodejs
[ 86%] Built target object_detection_generate_messages_cpp
[ 86%] Built target object_detection_generate_messages_py
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

Levi568, 2019-10-30:
The issue is resolved!

closed 2019-10-30 by Levi568

q8wwe, 2020-03-30:
how you solve it ??