leggedrobotics/darknet_ros #195: How can I use AlexeyAB version darknet,not the leggedrobotics'?

Shame-fight, 2019-11-05:
Thanks for your work,I want to use darknet on ros,the AlexeyAB'darknet is better than  leggedrobotics',But the code does not match the AB version.how can I use it,can you help me? the link is https://github.com/AlexeyAB/darknet 
Looking forward to your reply.
@ronwalf @alexmillane @ompugao @rmmcnulty9 @umer936

pushkalkatara, 2019-11-05:
Hi @Shame-fight Some time ago I did the AlexeyAB clone of darknet_ros. You can find it here - https://github.com/pushkalkatara/darknet_ros

Shame-fight, 2019-11-05:
> Hi @Shame-fight Some time ago I did the AlexeyAB clone of darknet_ros. You can find it here - https://github.com/pushkalkatara/darknet_ros

I am very happy to hear from you.In fact, I have tried your version on the PC, but I have a problem: when the target is detected, the process will be killed; even if the target is not detected, the same situation will occur after 1-2 minutes, I am not Know how to solve this problem.I look forward to using your code.

RhysMcK, 2019-11-28:
> Influenced from Memory Leak Issue on leggedrobotics/darket_ros repository. While running the ROS Node on Jetson TX1, the node was being killed in some time due to overflow in memory. This repository solves the issue of memory leak implementing multithreading techniques.

was that it?

Shame-fight, 2019-11-29:
> > Influenced from Memory Leak Issue on leggedrobotics/darket_ros repository. While running the ROS Node on Jetson TX1, the node was being killed in some time due to overflow in memory. This repository solves the issue of memory leak implementing multithreading techniques.
> 
> was that it?

sorry, I failed to run successfully, the program will be interrupted immediately when the target is detected. You can try to run this program normally.

tomlankhorst, 2021-06-08:
The package has changed a lot over time. Please feel free to open a new issue for current problems.

closed 2021-06-08 by tomlankhorst