leggedrobotics/darknet_ros #198: OpenCV 4.x+ requires enabled C++11 support

pedrobagatin, 2019-11-08:
> [ 43%] Building C object darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/activations.c.o
In file included from /usr/local/include/opencv4/opencv2/core/types_c.h:82:0,
                 from /usr/local/include/opencv4/opencv2/core/core_c.h:48,
                 from /usr/local/include/opencv4/opencv2/highgui/highgui_c.h:45,
                 from /home/pedro/catkin_ws/src/darknet_ros/darknet/include/darknet.h:25,
                 from /home/pedro/catkin_ws/src/darknet_ros/darknet/src/activations.h:3,
                 from /home/pedro/catkin_ws/src/darknet_ros/darknet/src/activation_layer.h:4,
                 from /home/pedro/catkin_ws/src/darknet_ros/darknet/src/activation_layer.c:1:
/usr/local/include/opencv4/opencv2/core/cvdef.h:654:4: error: #error "OpenCV 4.x+ requires enabled C++11 support"
 #  error "OpenCV 4.x+ requires enabled C++11 support"
    ^~~~~
compilation terminated due to -Wfatal-errors.

pedrobagatin, 2019-11-12:
F

Ajaygirish1, 2019-11-25:
@pedrobagatin is the problem solved?
if so, could you please tell me how you solved it.
Thank you

cxhmikegithub, 2020-08-06:
I meet the trouble too, could you tell me how to fix it ?

PrabhjotKaurGosal, 2020-08-14:
I am getting the same error. I am using ROS Noetic/Ubuntu 20.04.... I had the same problem when I was using ROS Melodic/Ubuntu 18.04

adamkrawczyk, 2020-08-29:
I also got this error

TohQiQi, 2020-09-09:
Is the problem solved? I also got this error.

ranjitkathiriya, 2020-11-29:
How to solve this error any idea Guys I am also getting the same error...

tomlankhorst, 2020-12-01:
See https://github.com/leggedrobotics/darknet_ros/issues/200

closed 2020-12-01 by tomlankhorst

bourrr, 2021-09-05:
I getting the same error, If someone solved it, please tell us how to overcome it.