leggedrobotics/darknet_ros #200: error "OpenCV 4.x+ requires enabled C++11 support"

Ajaygirish1, 2019-11-25:
Hey all,
I am facing the below error,
error "OpenCV 4.x+ requires enabled C++11 support" while executing catkin make,
I have tried inserting 
set(CMAKE_CXX_STANDARD 11)
set(CMAKE_CXX_STANDARD_REQUIRED TRUE)
in Cmake.txt of darknet_ros node but still doesnt work.

cardboardcode, 2019-12-05:
You can try have a look at **pull request #202** by @kunaltyagi. 

Compiled the version under this fork and it solved the compilation issue. 

Hope this helps.

mywnajsldkf, 2020-02-25:
which openCV version do you use??

kunaltyagi, 2020-02-25:
I use OpenCV@HEAD

mywnajsldkf, 2020-02-25:
@kunaltyagi oh! thank you for answering. I installed opencv3.4.6, after catkin_make darknet_ros, I met error "OpenCV 4.x+ requires enabled C++11 support" can I you your solution(https://github.com/leggedrobotics/darknet_ros/pull/202/commits/256ea5050a866dbd3973046428dcaeecc390a148 by following this commit) to solve my problem??

mywnajsldkf, 2020-02-25:
I want to use darknet_ros to detect something!! i

kunaltyagi, 2020-02-25:
> I installed opencv3.4.6, after catkin_make darknet_ros, I met error "OpenCV 4.x+ requires enabled C++11 support" can I you your solution(256ea50 by following this commit) to solve my problem??

You have OpenCV 4 not 3 installed, and yes, my fork works with latest OpenCV and ROS. If you want to work on darknet directly, I'd recommend https://github.com/alexeyab/darknet instead.

mywnajsldkf, 2020-02-25:
> You have OpenCV 4 not 3 installed, and yes, my fork works with latest OpenCV and ROS. If you want to work on darknet directly, I'd recommend https://github.com/alexeyab/darknet instead.

Oh, I'm sorry for bothering you
But when I enter `pkg_config --modeversion opencv` opencv3.4.6 show up. also I tested it on python. And I want to detect on ROS.

prit-ubuntu, 2020-07-15:
@Ajaygirish1 and @mywnajsldkf did you end up finding a solution? I am facing the same problems. I'm using Ubuntu 18.04 and OpenCV 3.4.0 for this purpose.

mywnajsldkf, 2020-07-15:
@prit-ubuntu Do you want to use OpenCV 3.4.0 for using darknet_ros? If you want it, I recommend following this procedure
1.  Delete OpenCV already installed your hardware! 
`sudo find / -name "*opencv*" -exec rm -i {} \;`
2. Install OpenCV (https://jkjung-avt.github.io/opencv-on-nano/)
i recommend this blog. And If you use ROS version1, ROS default python package is `python2.7`. So I changed shell script a little to install OpenCV 3.4.6 on python2

prit-ubuntu, 2020-07-17:
@mywnajsldkf so basically the solution you mentioned above worked for you to run darknet_ros? Yes, i do want to use OpenCV 3.4.0 for darknet_ros.

mywnajsldkf, 2020-07-18:
@prit-ubuntu It is best for me!

kunaltyagi, 2020-07-18:
You can also use my fork (I really should move it to point to AlexyAB's implementation)

prit-ubuntu, 2020-07-22:
@mywnajsldkf and @kunaltyagi I'm happy to inform you that I have finally managed to compile darknet_ros on my system! Thanks for your help.

Silerqin, 2020-09-09:
hi, I am getting the same issue. Can I know how to solve it? @kunaltyagi I have see your fork, can I know how to install it?

kunaltyagi, 2020-09-09:
There are other more up-to-date PRs. My fork has been unmaintained for a while so I personally wouldn't recommend it. See #257 and @230

However, my fork still works if you still want to use it. Please use the opencv4 branch, and clone it (with submodules) in your workspace (or underlay of that workspace) 😄

Silerqin, 2020-09-09:
@kunaltyagi I already use openCV4 but it still have error

tomlankhorst, 2020-12-01:
Support for OpenCV4 seems to be lagging a little. Maybe @mbjelonic has an indication of if and when it might come. 
@ranjitkathiriya

closed 2021-04-08 by mbjelonic