leggedrobotics/darknet_ros #204: Display BB only for few classes

pnambiar, 2020-01-08:
How to display BB only for few classes?

fiorano10, 2020-03-31:
What does your `.config` and `.yaml` file look like?

9527-csroad, 2022-10-17:
I also want to know if there are some methods to detect only one class or several classes? #158 says that we should to retarin a model, but I think there maybe a method to justify some place will make the job done.

9527-csroad, 2022-10-18:
I found a way to solve this question. 
In the YoloObjectDetector.cpp, I found this function
`draw_detections(display, dets, nboxes, demoThresh_, demoNames_, demoAlphabet_, demoClasses_);`
This function is used to draw BB and prob. So, just let it show the class that we want.
[image]
The modification in this picture is used to draw BB and prob of the first three classes (person, bicycle and car). 
If this method has any problems, let me know plz.