leggedrobotics/darknet_ros #205: exit code -11 while running darknet_ros.launch

ShiningDeadFishEyes, 2020-01-11:
error lookes below:

[darknet_ros-1] process has died [pid 23440, exit code -11, cmd /home/shanyx/catkin_ws/devel/lib/darknet_ros/darknet_ros camera/rgb/image_raw:=/image_raw __name:=darknet_ros __log:=/home/shanyx/.ros/log/5c4f5838-3449-11ea-8e87-54e1ad5a8195/darknet_ros-1.log].
log file: /home/shanyx/.ros/log/5c4f5838-3449-11ea-8e87-54e1ad5a8195/darknet_ros-1*.log

i have checked the logfile , nothing useful . Can anyone please help me ?

ShiningDeadFishEyes, 2020-01-13:
when i use gdb , ii got this :
```
[New Thread 0x7fff369fc700 (LWP 12457)]
Waiting for image.
[New Thread 0x7fff30dfe700 (LWP 12499)]

Thread 1 "darknet_ros" received signal SIGSEGV, Segmentation fault.
0x00007ffff7263e53 in ros::CallbackQueue::callAvailable(ros::WallDuration) ()
   from /opt/ros/lunar/lib/libroscpp.so

```

can anyone please help?

Scarf195, 2020-02-01:
Hi, same problem, did you fix that?

ShiningDeadFishEyes, 2020-02-09:
> Hi, same problem, did you fix that?

no,and i found something strange . When I used Ubuntu 16 in VMwrea I can conmpile and run perfectly (CPU version) , but when i install Ubuntu on my laptop with the same ISO , it shows errors . I have changed the kernel to the version i used in VMware .

ShiningDeadFishEyes, 2020-02-12:
> Hi, same problem, did you fix that?

Maybe i fix that . I reinstalled ubuntu16 today , but i didnt install boost or opencv , just ROS kinetic , now this project worked with CPU . I havent test the GPU version . Hope this is helpful to you .

ShiningDeadFishEyes, 2020-02-12:
> Hi, same problem, did you fix that?

Also remember to change the path of config and weights files in YoloObjectDetector.cpp .

Litao917, 2020-11-05:
Hey guys, I met the same issue. I'm on Ubuntu 18.04 with CUDA 10.2. 
Any clue about it? I currently would not like to reinstall my ROS as well as my Ubuntu. It is too costly for me. 
Many thanks!!!!

darkhan-s, 2021-04-03:
> Hey guys, I met the same issue. I'm on Ubuntu 18.04 with CUDA 10.2.
> Any clue about it? I currently would not like to reinstall my ROS as well as my Ubuntu. It is too costly for me.
> Many thanks!!!!

Hi, have you solved this problem?

tomlankhorst, 2021-06-08:
I hope a more recent version of darknet_ros solves this issue. If not, feel free to reopen this issue.

hanm2019, 2021-06-14:
I meet the same problem in melodic branch
```
![image](https://user-images.githubusercontent.com/56673361/121912603-4cdde900-cd63-11eb-8827-39e085647f73.png)

```

darkhan-s, 2021-06-14:
I think what I did back then to resolve this is that I retrained the model

tomlankhorst, 2021-06-14:
Maybe try to download the pre-trained models to be sure.

hanm2019, 2021-06-15:
> Maybe try to download the pre-trained models to be sure.

I try with pre-trained models , but meet the same error

dscroft, 2021-06-22:
Check the image format of the incoming message.
I was getting the same error but traced it to the rgbgr_image function when I was passing in mono images.
Switching to bgra8 caused the same exit code error but elsewhere.

Looks like only 3 channel image formats work but I'm still digging into it.

hanm2019, 2021-06-23:
> Check the image format of the incoming message.
> I was getting the same error but traced it to the rgbgr_image function when I was passing in mono images.
> Switching to bgra8 caused the same exit code error but elsewhere.
> 
> Looks like only 3 channel image formats work but I'm still digging into it.

I use ros_bridge to reform the image msg by:
```
cv_img_h = bridge.cv2_to_imgmsg(img_rot,"rgb8")
```
maybe  the image has 3 channel

TeingQiHao, 2021-11-26:
Loading weights from /home/bojan/darknet_ros_og/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolov4-tiny-custom_2000.weights...Done!
Waiting for image.
[darknet_ros-1] process has died [pid 13861, exit code -11, cmd /home/bojan/darknet_ros_og/devel/lib/darknet_ros/darknet_ros camera/rgb/image_raw:=/camera/rgb/image_raw __name:=darknet_ros __log:=/home/bojan/.ros/log/869f03e4-4e65-11ec-b140-78d0042abcfe/darknet_ros-1.log].
log file: /home/bojan/.ros/log/869f03e4-4e65-11ec-b140-78d0042abcfe/darknet_ros-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Hi everyone,

I am facing the same issue. I've trained a custom model on darknet yolov4-tiny. Followed the instructions on the readme.

1. Copied .weights file to darknet_ros/yolo_network_config/weights
2. Copied  .cfg file to darknet_ros/yolo_network_config/cfg
3. Changed ros.yaml, camera reading topic to /zed/zed_node/left/image_rect_color (using zed_ros_wrapper)
4. Changed yolov2-tiny.yaml to point to config files and weights file per 1. and 2.
5. Changed detection classes names in yolov2-tiny.yaml to the 1 class the model was trained on ("- box")

I was able to run the default yolov2-tiny.weights and .cfg but face this error only after pointing to our .cfg and weights as mentioned.

Please help.

MBouallegue, 2022-03-22:
This package helped me to use the yolov4 and worked with my custom dataset 
https://github.com/Tossy0423/darknet_ros/tree/020b7560f9c5901605503bb59450ed7654514e9c
I guess in the old version of darknet (melodic for me) some elements in the darknet/src is not updated to suits the yolov4

idochi, 2022-03-30:
I have the same problem and couldn't solve it yet. I'm using cv_camera for the image_raw. I'm new to Ros and need help.

Loading weights from /home/ez10/catkin_ws/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolov3.weights...Done!
Waiting for image.
[darknet_ros-1] process has died [pid 9663, exit code -11, cmd /home/ez10/catkin_ws/devel/lib/darknet_ros/darknet_ros camera/rgb/image_raw:=dev/video0 __name:=darknet_ros __log:=/home/ez10/.ros/log/eed6e908-aff5-11ec-8912-4f219ac33992/darknet_ros-1.log].
log file: /home/ez10/.ros/log/eed6e908-aff5-11ec-8912-4f219ac33992/darknet_ros-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

idochi, 2022-04-01:
The only thing, that worked for me was a new installation of Linux and ROS.

AhmetEkiz, 2023-06-07:
Hi guys, I have the same problem, but it also says:

**CUDA Error: no kernel image is available for execution on the device**

Before that, I deleted all `-gencode arch=compute_30,code=sm_30` like below on on [CMakeLists.txt](https://github.com/leggedrobotics/darknet_ros/blob/master/darknet_ros/CMakeLists.txt#L26) according to this [solution](https://github.com/leggedrobotics/darknet_ros/issues/363#issuecomment-985965791).

Then I change the line as a  `-gencode arch=compute_75,code=sm_75` according to this [solution](https://github.com/leggedrobotics/darknet_ros/issues/363#issuecomment-991019477). And It works. 

Also, I found a similar [solved issue](https://github.com/leggedrobotics/darknet_ros/issues/410#issue-1693724780).

### System (please complete the following information):

    OS: [Ubuntu 20.04]
    ROS version: [Noetic]
    GPU : [ GTX 1650]
    Cuda: 12.1
    Nvidia Driver Version: 530.30.02

`-gencode arch=compute_30,code=sm_30`
`-gencode arch=compute_35,code=sm_35`
`-gencode arch=compute_50,code=[sm_50,compute_50]`
`-gencode arch=compute_52,code=[sm_52,compute_52]`
`-gencode arch=compute_61,code=sm_61`
`-gencode arch=compute_62,code=sm_62`

Vamsi-IITI, 2023-07-09:
Try the following method : [Here](https://github.com/leggedrobotics/darknet_ros/issues/100#issuecomment-1627649963)