leggedrobotics/darknet_ros #212: image_interface.h:16:30: error: unknown type name ‘IplImage’ !!

JeyP4, 2020-02-15:
Hello
My system in ubuntu16.04 pc. 
ROS -kinetic. ROS installed its own `opencv-3.3.1-dev`. I compiled darknet with this opencv, and it working fine.

`catkin_make -DCATKIN_WHITELIST_PACKAGES="darknet_ros" -DCMAKE_BUILD_TYPE=Release`
But when I try to buid darknet_ros, it is throwing below error. I noticed that no `OpenCV found` line in building.
[image]
**How this error can be resolved?**

thewayofstome, 2020-03-14:
i have same problem.

chillcloud-dev, 2020-03-17:
i have same problem.

wzmzero, 2020-03-22:
i also have same problem.

arnaldojr, 2020-04-17:
same here too...

tomlankhorst, 2020-04-17:
On Docker `ros:kinetic`:

```bash
apt-get update && apt-get install python-catkin-tools
mkdir -p ws/src && cd ws
git clone --recursive https://github.com/leggedrobotics/darknet_ros.git src/darknet_ros
catkin init
rosdep install -y --from-paths src
catkin build -DCMAKE_BUILD_TYPE=Release
```
...succeeds without issues. 

That's with
- libopencv-dev: 2.4.9.1+dfsg-1.5ubuntu1.1

Please provide a minimal example that reproduces this issue.

jcmayoral, 2020-04-17:
Hey. It is working for me using opencv  3.2.0-rc, After some test from 3.3 or above seems no to be working.

arnaldojr, 2020-04-20:
I think that my problem was openCV4. I reinstall my system with opencv3 and worked.

fwarmuth, 2020-05-07:
same problem here:
but i am using a Jetson Xavier using OpenCV 3.3.1...
any clues?

scosar, 2020-12-18:
> same problem here:
> but i am using a Jetson Xavier with OpenCV 3.3.1...
> any clues?
> i will try to setup a Docker container as suggested

Hello, any luck with this? 

I am also using Jetson Xavier. I do not want to uninstall OpenCV 4. I compiled OpenCV 3.4.12 from source. I set the OpenCV_DIR in CMakeLists.txt to refer to the compiled version:

`set(OpenCV_DIR /usr/local/share/OpenCV)`

However, it is still giving the same error.

hasanzfdn, 2021-02-24:
I have same problem in ros melodic. Did anyone solve this problem ?

arnaldojr, 2021-02-24:
> I have same problem in ros melodic. Did anyone solve this problem ?

Are you using OpenCV 4? Try reinstall the opencv3.

hasanzfdn, 2021-02-24:
> > I have same problem in ros melodic. Did anyone solve this problem ?
> 
> Are you using OpenCV 4? Try reinstall the opencv3.

thank you. I tried it but wasnt work too. 
https://github.com/ShiSanChuan/darknet_ros  I downloaded this repo. and its working now .