leggedrobotics/darknet_ros #214: Question:OpenCV version

mywnajsldkf, 2020-02-19:
Before `build` darknet_ros, we have to install Opencv, which version is best suit for darknet_ros??

stevenlee090, 2020-03-10:
I am wondering about the same question too. I am trying to use this with a Nvidia Jetson Nano board which has a system image with OpenCV 4 installed. But that doesn't seem to work well with `darknet` library.

As per discussion [here](https://github.com/pjreddie/darknet/issues/1754), it might be difficult to resolve this. Did you eventually get it working? If so, which version of OpenCV did you use?

mywnajsldkf, 2020-03-10:
@stevenlee090 
Oh! I really struggled to solve that too, haha
First of all, I deleted opencv that Jetson Nano and `install opencv3.4.6` by following this page
https://jkjung-avt.github.io/opencv-on-nano/
It is so easy! Because the writer made shell file already
But Keep in mind. You have to change `shell` a little to download opencv3.4.6 on python2

I hope it will be helpful.

stevenlee090, 2020-03-10:
@mywnajsldkf 
Thanks for the prompt feedback!
I will try that out and see how it goes.

closed 2021-06-08 by tomlankhorst