leggedrobotics/darknet_ros #216: error: #error "OpenCV 4.x+ requires enabled C++11 support"

mywnajsldkf, 2020-02-26:
```
[ 41%] Built target darknet_ros_msgs_generate_messages
[ 41%] Building C object darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/src/image_interface.c.o
[ 42%] Building C object darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/activation_layer.c.o
[ 42%] Building C object darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/activations.c.o
[ 43%] Building C object darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/__/darknet/src/image.c.o
In file included from /usr/include/opencv/opencv2/core/types_c.h:82:0,
                 from /usr/include/opencv/opencv2/core/core_c.h:48,
                 from /usr/include/opencv/opencv2/highgui/highgui_c.h:45,
                 from /home/nano2/catkin_ws/src/darknet_ros/darknet/include/darknet.h:25,
                 from /home/nano2/catkin_ws/src/darknet_ros/darknet/src/box.h:3,
                 from /home/nano2/catkin_ws/src/darknet_ros/darknet/src/image.h:9,
                 from /home/nano2/catkin_ws/src/darknet_ros/darknet_ros/include/darknet_ros/image_interface.h:12,
                 from /home/nano2/catkin_ws/src/darknet_ros/darknet_ros/src/image_interface.c:9:
/usr/include/opencv/opencv2/core/cvdef.h:690:4: error: #error "OpenCV 4.x+ requires enabled C++11 support"
```

hi! I have an error during `catkin_make -DCMAKE_BUILD_TYPE=Release` 
I installed opencv3.4.6 and I checked it python3 and pkg-confing --modversion opencv
how can I solve it??

mywnajsldkf, 2020-02-26:
I finally solved...
I checked my jetson-nano
There is some folders named "opencv4" 
I deleted them and it was solved
But 
`pkg-config --modversion opencv4`
`4.1.1` show up! haha 
and I meet an anoter error...

closed 2020-03-05 by mywnajsldkf

gloria201, 2020-10-22:
Hi, #error "OpenCV 4.x+ requires enabled C++11 support" occurs when i built the package. I just automatically install opencv4. Is this issue able to be solved if I downgrade it completely to opencv 3.4.6?

gloria201, 2020-10-22:
Hi, @mywnajsldkf  
#error "OpenCV 4.x+ requires enabled C++11 support" occurs when i built the package. I just automatically install opencv4. Is this issue able to be solved if I downgrade it completely to opencv 3.4.6?

mywnajsldkf, 2020-10-22:
Hi! @gloria201
Actually, I'm not sure about that. But other methods to install opencv 3.4.x said 
1. delete all of opencv installed in your PC
2. And reinstall opencv 3.4.x 

Have a nice coding!!!!