leggedrobotics/darknet_ros #218: Problems with Intel Real Sense D435i

Philipp385, 2020-03-02:
Hello,

i wanted to start detecting custom objects with my Intel Real Sense and trained weights from YoloV3. Therefore i installed the IR Package from https://github.com/IntelRealSense/realsense-ros
and the darknet-ros package from here (https://github.com/leggedrobotics/darknet_ros).
I have already trained yolo on a custom dataset and put the weight-file in the weights folder and the cfg-file in the cfg folder. Then I edited the ros.yaml file to the Topic "/camera/color/image_raw" from the IntelRealSense and generated a yaml file for my custom objects as you can see here:
[image]
I edited the darknet_ros launch file in line 14 to point to my yaml file:
[image]
In the next step I started (each in an extra terminal) the roscore, IntelRealSense Node (roslaunch realsense2_camera rs_camera.launch) and the darknet ros launch file (roslaunch darknet_ros darknet_ros.launch) and i get the result:
[image]
But after a time i get the following error:
[image]

Has anyone an idea what causes this problem and could please help me to find a solution? Is there a mistake i made?

Thank you in advance.

Philipp385, 2020-03-03:
I solved the Problem. It was the sharp S in "Stator groß", that caused the problem. Everytime YOLO detected the "Stator groß" all processes died.

closed 2020-03-03 by Philipp385

hsaleem1, 2020-10-02:
@Philipp385,
Please help me.
I have installed the package darknet_ros
Now I ran roscore, Intel Realsense node and darknet_ros nodes.
What should I do next? should I type rviz in the other terminal or what not?