leggedrobotics/darknet_ros #221: Using a PC with a GPU, but I have only 1.3 FPS

sayami888, 2020-03-17:
I use PC with GeForce RTX 2070,but I have only 1.3 FPS.
I am trying darknet_ros using a USB camera. The settings for the USB camera worked fine(usb_cam).
Editing:
 darknet_ros/darknet_ros/config/ros.yaml   topic: /camera/rgb/image_raw   ->  /usb_cam/image_raw 
 darknet_ros/darknet/Makefile                     GPU=0  OPENCV=0                 -> GPU=1  OPENCV=1
Building again,execuiting "$ roslaunch darknet_ros darknet_ros.launch",and video is displaying and YOLO is working, but FPS is 1.3.
I have been using darknet for a long time,but I am not very familiar with darknet_ros.
I seem that the GPU is not used.

Sorry. Please help me.

Dimitrios-Ar, 2020-03-17:
Hi. One way of checking your gpu is by typing "nvidia-smi -l" and check the changes of memory when running darknet. Additionally, you can try running darknet and checking the difference when having GPU = 1  and GPU = 0 in the Makefile. 

I had a similar issue which was caused by using a non-official charger for my PC forcing my GPU to function slowly because it was detecting a small voltage difference. When the charger is not plugged. the fps goes back to normal. I guess it won't be the case for you but I mentioned it just in case.

sayami888, 2020-03-18:
@Dimitrios-Ar 
Thank you for answering.
After looking again about GPU,I thought there was something wrong with CUDA.
I have not solved the problem yet, but I have a clue to the solution.Thank you.

AndreasGronlund, 2020-03-20:
I have the same issue. Darknet works as it should with cuda but when applying the darknet_ros layer the fps is really low (0.2fps). Any progress so far @sayami888 ?

sayami888, 2020-03-23:
@AndreasGronlund 
Sorry,I have not solved the problem yet.
I would like someone to tell me,do I need to do "make" in darknet_ros/darknet?
There is an error when I do "make" it, so that may be the cause of the problem.

nvcc fatal   : Path to libdevice library not specified
Makefile:88: recipe for target 'obj/convolutional_kernels.o' failed
make: *** [obj/convolutional_kernels.o] Error 1

AndreasGronlund, 2020-03-30:
@sayami888 The problem for me is in the operation `detect_thread.join();` on line 554 in YoloObjectDetector.cpp which consumes a lot of time (for me ~6 sec per iteration and being close to 100% of the computational time). Why it is consuming so much time I don't know.

fiorano10, 2020-03-31:
Are you sure the prediction is running on the GPU, and not your CPU? Maybe look at the utilization for both.

sayami888, 2020-03-31:
When predicting with TinyYOLOv3 using a USB camera on a normal darknet,Volatile GPU-Util is 22%,GPU Memory Usage is 395MiB.
When predicting with darknet_ros and YOLOv3,Volatile GPU-Util is 3%,GPU Memory Usage is 97MiB.
This is definitely not using a GPU.

tomlankhorst, 2020-06-17:
Seems like CUDA is not found.
Check the output of CMake when building `darknet_ros`. 
You should see a `CUDA Version` output when CUDA is found. 
https://github.com/leggedrobotics/darknet_ros/blob/master/darknet_ros/CMakeLists.txt#L20

closed 2021-05-27 by tomlankhorst