leggedrobotics/darknet_ros #222: CMakeList.txt catkin_make error undefined reference openCV

lchop, 2020-03-18:
On ROS kinetic 16.04  latest commit made catkin_make not possible to build the package:
 undefined reference to `cv::resizeWindow(cv::String const&, int, int)'

I suspect commit: c30619a5b5432f62205482973e87afacb1499d87 from @kunaltyagi 

I went back to commit 42e31571ed257d7d0eb1c069892ab868dbe299dc and it is perfectly working and compiling. 

For now I have not identified the error.

kunaltyagi, 2020-03-18:
`cv::resizeWindow(cv::String const&, int, int)` is present all the way back in [v2.4](https://docs.opencv.org/ref/2.4/d2/d75/namespacecv.html#a59d5acf3eb6c905321426f1b0d36678c)

lchop, 2020-03-20:
Yes I know but it is not the error. It is a linking error due to the cleaning of the CMakeLists.txt, I guess. I have just point this undefined reference but I have plenty more in the error. And when I go back just from one commit behind it is perfectly working without changing anything.

kunaltyagi, 2020-03-22:
> It is a linking error due to the cleaning of the CMakeLists.txt

Got it. Could you provide the following 2:
* CMake config (output of building a package from scratch, before the building/linking prompts with percentage)
* Verbose build/link commands that fail (using `VERBOSE=1 <command used to build>` and finding the lines just above the error)

jorgemia, 2020-03-27:
@kunaltyagi Hi I was having some issues with catkin make after I installed your files from the pull request to make darknet ros compatible with open cv 4. After changing 

```
if (CUDA_FOUND)
  link_directories(
    #${CUDA_TOOLKIT_ROOT_DIR}/lib
  )
```
to 

```
if (CUDA_FOUND)
  link_directories(
    /usr/local/cuda/lib64
  )
```
it worked.

This was the error I was getting:

```
/usr/bin/ld: cannot find -lcudart
/usr/bin/ld: cannot find -lcublas
/usr/bin/ld: cannot find -lcurand
collect2: error: ld returned 1 exit status
darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:1961: recipe for target '/home/jorge/catkin_ws/devel/lib/libdarknet_ros_lib.so' failed
make[2]: *** [/home/jorge/catkin_ws/devel/lib/libdarknet_ros_lib.so] Error 1
CMakeFiles/Makefile2:3141: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

```

kunaltyagi, 2020-03-27:
>  #${CUDA_TOOLKIT_ROOT_DIR}/lib

That's a mistake if so. It should be ` ${CUDA_TOOLKIT_ROOT_DIR}/lib` (no hash/pound)

jorgemia, 2020-03-27:
Yeah sorry that’s a copy from my code where I commented out that line in my file

kunaltyagi, 2020-03-27:
This is a wrong place to discuss this particular error, but old cmake versions (or bad findCuda modules) can result in such errors. Hard-coding the location of cuda is harmful for the generic user.

closed 2021-05-19 by tomlankhorst