leggedrobotics/darknet_ros #223: error: unknown type name ‘IplImage’; did you mean ‘image’?  void generate_image(image p, IplImage *disp);

wzmzero, 2020-03-22:


fwarmuth, 2020-05-07:
any solution?

annb3, 2020-11-17:
Same problem!

VbsmRobotic, 2021-06-02:
Hi, I also have same problem. Is there any solution?

VbsmRobotic, 2021-06-02:
change Makefile :
ifeq ($(OPENCV), 1) 
COMMON+= -DOPENCV
CFLAGS+= -DOPENCV
LDFLAGS+= `pkg-config --libs opencv4` -lstdc++
COMMON+= `pkg-config --cflags opencv4` 
endif

src/darknet_ros/darknet_ros/include/darknet_ros/image_interface.h:16:30: error: unknown type name ‘IplImage’; did you mean ‘image’?
 void generate_image(image p, IplImage* disp);
Solution:
line 16 add : 
#ifdef OPENCV4
void generate_image(image p, IplImage* disp);

src/darknet_ros/darknet_ros/include/darknet_ros/YoloObjectDetector.hpp:62:17: error: conflicting declaration of C function ‘void show_image_cv(image, const char*, IplImage*)’
 extern "C" void show_image_cv(image p, const char* name, IplImage* disp);
Solution:
line62 add:
#ifdef OPENCV4
extern "C" void show_image_cv(image p, const char* name, IplImage* disp);
#endif

/src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:383:57: error: invalid conversion from ‘IplImage* {aka _IplImage*}’ to ‘int’ [-fpermissive]
   show_image_cv(buff_[(buffIndex_ + 1) % 3], "YOLO V3", ipl_);
Solution:
  #ifdef OPENCV4
  show_image_cv(buff_[(buffIndex_ + 1) % 3], "YOLO V3", ipl_);
  #endif

/src/darknet_ros/darknet/examples/attention.c:303:13: error: too few arguments to function ‘show_image’
             show_image(tile, "tile");
Solution:
#ifdef OPENCV4
            show_image(tile, "tile");
#endif

src/darknet_ros/darknet_ros/src/YoloObjectDetector.cpp:500:9: error: ‘generate_image’ was not declared in this scope
         generate_image(buff_[(buffIndex_ + 1) % 3], ipl_);
Solution:
#ifdef OPENCV4
        generate_image(buff_[(buffIndex_ + 1) % 3], ipl_);
#endif
I have changed and make it 
[image]

tomlankhorst, 2021-06-08:
I hope this issue was resolved with the latest release that uses OpenCV4 instead of OpenCV3.
@VbsmRobotic If you are still facing issues I would like to invite you to open a new issue and provide all the details of your problem there. That way, it will be easier for people to help you. Thanks!

closed 2021-06-08 by tomlankhorst

SDG6199, 2021-07-29:
Isn't the compilation incomplete?

How do I prevent 
'recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all' failedrecipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros.dir/all' failed'
from occurring?