leggedrobotics/darknet_ros #224: Face is detected but bounding box is not drawn and not published

lchop, 2020-03-24:
Hello,

I have follow the indication to include a custom training for YOLO:

I include the cfg file and the weight file from: https://github.com/dannyblueliu/YOLO-Face-detection
I change the config file as:

yolo_model:

  config_file:
    name: yoloface.cfg
  weight_file:
    name: yoloface.weights
  threshold:
    value: 0.3
  detection_classes:
    names:
      - face
When I launch everything seems to work, the fps is high and the percentage of the face detected is displayed but not the bouding box and also the coordinates are not published. Could you help me ?

fiorano10, 2020-03-31:
Did you change the batch size, and subdivision to 1?

Shreyas0812, 2020-07-20:
Sir, I am sorry, this is not about the issue you have raised. I am unable to do custom detection at all. I am really new to ros and yolo. Would you please guide me in what you did to achieve the same. it would be really great:)

Thank you!