leggedrobotics/darknet_ros #225: catkin make fails jetson nano opencv 4.0

jorgemia, 2020-03-27:
I'm working on a Jetson Nano which comes with openCV 4.0, so tried to get darknet_ros working by using the changes in PR #202 . When running catkin make I get the following errors:

```
/usr/bin/ld: cannot find -lcudart
/usr/bin/ld: cannot find -lcublas
/usr/bin/ld: cannot find -lcurand
collect2: error: ld returned 1 exit status
darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/build.make:1961: recipe for target '/home/jorge/catkin_ws/devel/lib/libdarknet_ros_lib.so' failed
make[2]: *** [/home/jorge/catkin_ws/devel/lib/libdarknet_ros_lib.so] Error 1
CMakeFiles/Makefile2:3141: recipe for target 'darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all' failed
make[1]: *** [darknet_ros/darknet_ros/CMakeFiles/darknet_ros_lib.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```
I hard coded the location of cuda in the CMakeLists file which solved the problem, but @kunaltyagi advised against this. Any ideas how to go about fixing this?

marcostrullato, 2020-10-18:
Same issue

Sunshengjin, 2021-01-26:
Facing same problem

closed 2021-04-08 by mbjelonic