leggedrobotics/darknet_ros #227: 0 CUDA Error: unknown error

renatojmsdh, 2020-03-31:
Did anyone find a solution? 

I'm having the same issue my PC. Neither the weights or configuration are loaded.

```
[ INFO] [1585614661.644136477]: [YoloObjectDetector] Node started.
[ INFO] [1585614661.680486156]: [YoloObjectDetector] Xserver is running.
[ INFO] [1585614661.681623332]: [YoloObjectDetector] init().
YOLO V3
layer     filters    size              input                output
    0 CUDA Error: unknown error
CUDA Error: unknown error: Success
```
@fiorano10 my **nvcc -V:**

```
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2018 NVIDIA Corporation
Built on Sat_Aug_25_21:08:01_CDT_2018
Cuda compilation tools, release 10.0, V10.0.130

```

_Originally posted by @1329844541 in https://github.com/leggedrobotics/darknet_ros/issues/172#issuecomment-513744713_

fiorano10, 2020-03-31:
did you clone using `--recursive` when cloning the repo?

renatojmsdh, 2020-03-31:
Thanks for the answer @fiorano10 . Yup I did use recursive.

I just succeed solving now what was going on. So for reference: I started trying the darknet standalone project, which provided a more specific error:
`darknet: ./src/cuda.c:36: check_error: Assertion `0' failed`
I though then it could be a memory issue (my GPU has only 4GB), but both batch and minisections are 1 in the tested .cfg nets. After looking around for a while I decided to check if all my drivers and cuda installs were working properly with a STRONG sanity check of the GPU install: **tensorflow**. After doing **tf.test.is_gpu_available()** I got the message that the NVIDIA driver **kernel version does not match DSO version**.  So I reinstalled the cuda-10.0 package including the drivers and things start working!

closed 2020-04-01 by renatojmsdh