leggedrobotics/darknet_ros #234: Installation Issues (Opencv 4.x requires enabled C++11 support)

zulfiz, 2020-04-24:
I am getting the following message when I try to build the code:
`
/usr/include/opencv4/opencv2/core/cvdef.h:690:4: error: #error OpenCV 4.x+ requires enabled C++11 support error OpenCV 4.x+ requires enabled C++11 support ^~~~~ compilation terminated due to -Wfatal-errors.
`
Is there a way around this issue? Thank you!

ShiningDeadFishEyes, 2020-05-07:
opencv 3.4.0 works .  also if you use opencv3.4.0,you should use an older version of darknet , link:https://gitee.com/shanyixiang/darknet_ros_qqqqqqq , just replace the darknet folder .

NerminSalem, 2020-05-09:
@zulfiz i got the same error, did you solve it?

plugns, 2020-08-02:
i got the same error, did you solve it?

tomlankhorst, 2020-11-29:
The suggested fix is to use OpenCV 3.4, which is the default on Ubuntu Bionic.

closed 2020-11-29 by tomlankhorst