leggedrobotics/darknet_ros #235: Why use SSH to clone this repository?

willzoe, 2020-04-27:
I used the `git clone` command to clone this repository by `HTTPS` to my catkin workspace, and an error occurred during compilation.

Later, I followed the prompts in the `README` and used `SSH` to clone this repository. The package can be compiled correctly.

I want to know what is the difference between using HTTP cloning and SSH cloning?

tomlankhorst, 2020-04-28:
There should be no difference. Could you have missed the  `--recursive` flag the first time when you were cloning with HTTPS?

willzoe, 2020-04-28:
I did miss this parameter. 

What is the role of this parameter? If this parameter is not used, will the repository be missing some files?

tomlankhorst, 2020-04-28:
This repository contains a [submodule](https://git-scm.com/book/en/v2/Git-Tools-Submodules), that's like another repository in this repository. The `--recursive` flags indicates that `git` should clone these sub repositories too.

closed 2020-04-28 by tomlankhorst