leggedrobotics/darknet_ros #236: How to start detection only when need to detect?

willzoe, 2020-04-27:
I use TX2 to control the robot to complete SLAM and navigation. During the navigation process, I need to detect several frames of images in a short period of time (about 2 seconds) at a specified location. 

In order to save the computing resources of TX2, I want to suspend detect during other non-detect time. But I found that when I start the darknet_ros node, the node keeps detecting the images collected by the camera.

What should I do to detect the image captured by the camera only when needed?

sachinkum0009, 2020-05-08:
Hi, I am not an expert. But you can try to remap that topic to something else when you don't want darknet_ros to detect object and when you want it to detect again remap the topic to the required one.
If it work or not, please comment down.

alexis-pepe, 2020-08-21:
hello, I am also using TX2 to run Darknet_ros, in the makefile, I edited  GPU=1 OPENCV=1 CUDNN=1, combined with Yolov3.Weights ,FPS is only about 0.7, When I use Yolov2-tiny weights, FPS increases to about 3.8FPS.

I'm so confused, because I see on the Internet that yolov2-Tiny can reach 20+ FPS. Is my TX2 configuration wrong? When I run "sudo jetson_release", output the following:
 - NVIDIA Jetson TX2
   * Jetpack 3.0 [L4T 27.1.0]
   * NV Power Mode: MAXN - Type: 0
   * jetson_stats.service: active
 - Libraries:
   * CUDA: 8.0.64
   * cuDNN: 5.1.10
   * TensorRT: NOT_INSTALLED
   * Visionworks: 1.6.0.233n
   * OpenCV: 3.3.0 compiled CUDA: YES
   * VPI: NOT_INSTALLED
   * Vulkan: 
Is there a problem here? Could it be the version that caused the problem?

Can you tell me how your CUDNN is configured and whether this path exists in your files
My /usr folder does not have this path"/usr/local/cudnn/". Is this correct? In the /usr/local/cuda/include ,I can find cudnn.h, in the /usr/local/cuda/lib64/ ,I can find some file like "libcudnn.so"

This problem has been bothering me for many days, and I hope to get your guidance, Thank you! @willzoe

willzoe, 2020-08-27:
After using SDK Manager to configure TX2, I haven't configured cuDNN again.

The FPS is very slow may be because the input video resolution is too high. You can try to adjust the resolution of the input image to 640*480 or 1280*720.