leggedrobotics/darknet_ros #238: I can't solve this problem by all ways I tried , Help me !!! [ERROR: cannot launch node of type [darknet_ros/darknet_ros]: Cannot locate node of type [darknet_ros] in package [darknet_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)]

Eslamrady42, 2020-05-07:
root-@root:~$ roslaunch darknet_ros yolo_v3.launch 
... logging to /home/root-/.ros/log/5d4475c2-904a-11ea-a70d-dca2664ca9fb/roslaunch-root-14265.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://root:38005/

SUMMARY
========

PARAMETERS
 * /darknet_ros/actions/camera_reading/name: /darknet_ros/chec...
 * /darknet_ros/config_path: /home/root-/catki...
 * /darknet_ros/image_view/enable_console_output: True
 * /darknet_ros/image_view/enable_opencv: True
 * /darknet_ros/image_view/wait_key_delay: 1
 * /darknet_ros/publishers/bounding_boxes/latch: False
 * /darknet_ros/publishers/bounding_boxes/queue_size: 1
 * /darknet_ros/publishers/bounding_boxes/topic: /darknet_ros/boun...
 * /darknet_ros/publishers/detection_image/latch: True
 * /darknet_ros/publishers/detection_image/queue_size: 1
 * /darknet_ros/publishers/detection_image/topic: /darknet_ros/dete...
 * /darknet_ros/publishers/object_detector/latch: False
 * /darknet_ros/publishers/object_detector/queue_size: 1
 * /darknet_ros/publishers/object_detector/topic: /darknet_ros/foun...
 * /darknet_ros/subscribers/camera_reading/queue_size: 1
 * /darknet_ros/subscribers/camera_reading/topic: /usb_cam/image_raw
 * /darknet_ros/weights_path: /home/root-/catki...
 * /darknet_ros/yolo_model/config_file/name: yolov3.cfg
 * /darknet_ros/yolo_model/detection_classes/names: ['person', 'bicyc...
 * /darknet_ros/yolo_model/threshold/value: 0.3
 * /darknet_ros/yolo_model/weight_file/name: yolov3.weights
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
    darknet_ros (darknet_ros/darknet_ros)

auto-starting new master
process[master]: started with pid [14275]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5d4475c2-904a-11ea-a70d-dca2664ca9fb
process[rosout-1]: started with pid [14286]
started core service [/rosout]
ERROR: cannot launch node of type [darknet_ros/darknet_ros]: Cannot locate node of type [darknet_ros] in package [darknet_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
root-@root:~$

Eslamrady42, 2020-06-11:
Solved

closed 2020-06-11 by Eslamrady42

tomlankhorst, 2020-06-16:
Would you mind sharing the solution to this problem that you couldn't solve in all ways you tried? Thanks in advance!

Eslamrady42, 2020-06-17:
Sure, first of all remove everything that related to darknet on your pc expat the Weights of yolo if you downloaded it already. 
Start from beginning from this link:
https://github.com/leggedrobotics/darknet_ros

The **Building** part you need do it by SSH method by this link mentioned before in the article:
https://confluence.atlassian.com/bitbucket/set-up-an-ssh-key-728138079.html

Then continue the other instructions without any errors or problems.

tomlankhorst, 2020-06-17:
It seems you did not have access to a required repository. Thank you for your response and all the best.

Eslamrady42, 2020-06-17:
You’re welcome.