leggedrobotics/darknet_ros #239: cuda : out of memory

liangpu597, 2020-05-08:
I have troubles:
`CUDA Error: out of memory
CUDA Error: out of memory: File exists
[darknet_ros-2] process has died [pid 1669, exit code 255, cmd /home/tony/catkin_ws/devel/lib/darknet_ros/darknet_ros camera/rgb/image_raw:=/gi/simulation/left/image_raw __name:=darknet_ros __log:=/home/tony/.ros/log/3efd60c0-90fd-11ea-9b0a-b82a72b973ed/darknet_ros-2.log].
log file: /home/tony/.ros/log/3efd60c0-90fd-11ea-9b0a-b82a72b973ed/darknet_ros-2*.log`

please telle me how to solve it? thanks a lot!

tomlankhorst, 2020-06-03:
Post hardware specs, OS, commit, etc., etc.

tomlankhorst, 2020-06-17:
The error indicates that you do not have enough memory available on your GPU. 

- Reduce GPU memory usage by other programs (check with `nvidia-smi`)
- Reduce GPU memory usage by Darknet (use different weights; smaller images)
- Use a GPU with more memory

closed 2020-06-17 by tomlankhorst