leggedrobotics/darknet_ros #242: Is it possible to use a protobuf file (.pb) instead of .weights file?

dregner, 2020-05-18:
Hello,
I am developing a detection using my own training files to detect some specific objects, unfortunately I trained using darkflow. Is it possible to use a .pb file instead a .weights in darknet_ros? 
I have been looking all around internet for a way to convert from .pb to .weights but do not get a strict answer how I can do this.

tomlankhorst, 2020-06-16:
Protobuf is a serialization mechanism, you can put any data you like into it. 
You will probably have to write your own procedure to convert your protobuf file to a weights file if you don't have any more information on how you structured your .pb.

dregner, 2020-06-16:
Thanks for your answer.

At the end I found some ways on internet how to convert from protobuf to a weights file, but all was really confuse and messy. So I preferred to train the neural network using darknet instead of darkflow and it gives me the file I needed easily than try to convert.

closed 2020-06-16 by dregner

tomlankhorst, 2020-06-17:
That's great! If you have any resources you would like to share for other people to look at and learn from, feel free to add a comment here. Thanks for your time.

dregner, 2020-06-17:
[here](https://github.com/thtrieu/darkflow/issues/811) and [here](https://github.com/thtrieu/darkflow/issues/198) are some information to convert