leggedrobotics/darknet_ros #243: will yolo v4 be supported?

Sunchanghao, 2020-05-19:


zclonely, 2020-05-31:
Do you mean that only v2, v3, v3-tiny are supported now?
I encountered a bug when implement v3-spp and v4, I don’t know how to solve it, but the same operation will not appear on v2, v3, v3-tiny
[image]

makangzhe, 2020-06-02:
https://github.com/tom13133/darknet_ros   This may help you

Shame-fight, 2020-06-03:
> https://github.com/tom13133/darknet_ros This may help you

Thank you, this is good news. Have you tried this version? On PC or jetson-tx2 machines, can it be compiled and passed.

Shame-fight, 2020-06-03:
I successfully ran this version of the code on the PC, which was very pleasing. But I ran into a problem: the exact same bounding box was posted twice. As far as I know, the previous version of this repository also had the same problem. At the same time, I am not sure if there is a continuous increase in memory, because I only ran for two minutes. Welcome to discuss.
@diegocdl @ganechi @zclonely @Sunchanghao @

makangzhe, 2020-06-03:
> I successfully ran this version of the code on the PC, which was very pleasing. But I ran into a problem: the exact same bounding box was posted twice. As far as I know, the previous version of this [...]

Do you know some methods applied for ROS of 6D pose estimation in github?

Shame-fight, 2020-06-04:
> 6D pose

[https://zhuanlan.zhihu.com/p/94020758](url)
this may help you

Shame-fight, 2020-06-05:
I have encountered a problem, ros has published the topic of /darknet_ros/boundingoxes, how to write code to subscribe to this topic and use it for future navigation?how did you solve it

jorgemia, 2020-06-17:
> > https://github.com/tom13133/darknet_ros This may help you
> 
> Thank you, this is good news. Have you tried this version? On PC or jetson-tx2 machines, can it be compiled and passed.

This fork seems to work on a Jetson Xavier NX but I'm only getting around 3FPS. When I built using catkin_make I got a lot of warnings similar to things like:

```
/home/unmnd/catkin_ws/src/darknet_ros/darknet/src/activations.c: In function ‘activate’:
/home/unmnd/catkin_ws/src/darknet_ros/darknet/src/activations.c:77:5: warning: enumeration value ‘RELU6’ not handled in switch [-Wswitch]
     switch(a){
     ^~~~~~
/home/unmnd/catkin_ws/src/darknet_ros/darknet/src/activations.c:77:5: warning: enumeration value ‘SWISH’ not handled in switch [-Wswitch]
/home/unmnd/catkin_ws/src/darknet_ros/darknet/src/activations.c:77:5: warning: enumeration value ‘MISH’ not handled in switch [-Wswitch]
```

or

```
/home/unmnd/catkin_ws/src/darknet_ros/darknet/src/gaussian_yolo_layer.c: In function ‘make_gaussian_yolo_layer’:
/home/unmnd/catkin_ws/src/darknet_ros/darknet/src/gaussian_yolo_layer.c:71:38: warning: passing argument 1 of ‘cudaHostAlloc’ from incompatible pointer type [-Wincompatible-pointer-types]
     if (cudaSuccess == cudaHostAlloc(&l.output, batch*l.outputs * sizeof(float), cudaHostRegisterMapped)) l.output_pinned = 1;
                                      ^
In file included from /usr/local/cuda/include/cuda_runtime.h:96:0,
                 from /home/unmnd/catkin_ws/src/darknet_ros/darknet/include/darknet.h:41,
                 from /home/unmnd/catkin_ws/src/darknet_ros/darknet/src/gaussian_yolo_layer.h:5,
                 from /home/unmnd/catkin_ws/src/darknet_ros/darknet/src/gaussian_yolo_layer.c:7:
/usr/local/cuda/include/cuda_runtime_api.h:4575:39: note: expected ‘void **’ but argument is of type ‘float **’
 extern __host__ cudaError_t CUDARTAPI cudaHostAlloc(void **pHost, size_t size, unsigned int flags);
                                       ^~~~~~~~~~~~~
/home/unmnd/catkin_ws/src/darknet_ros/darknet/src/gaussian_yolo_layer.c:78:38: warning: passing argument 1 of ‘cudaHostAlloc’ from incompatible pointer type [-Wincompatible-pointer-types]
     if (cudaSuccess == cudaHostAlloc(&l.delta, batch*l.outputs * sizeof(float), cudaHostRegisterMapped)) l.delta_pinned = 1;
                                      ^

```
So it works but not sure if it built correctly. This is what I get when I launch the node:

```
auto-starting new master
process[master]: started with pid [25905]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to cf0eb048-b0e6-11ea-ac5b-70665563d9f5
process[rosout-1]: started with pid [25916]
started core service [/rosout]
process[darknet_ros-2]: started with pid [25923]
[ INFO] [1592431599.098680603]: [YoloObjectDetector] Node started.
[ INFO] [1592431599.110482161]: [YoloObjectDetector] Xserver is running.
[ INFO] [1592431599.113667653]: [YoloObjectDetector] init().
YOLO
 Try to load cfg: /home/unmnd/catkin_ws/src/darknet_ros/darknet_ros/yolo_network_config/cfg/yolov4.cfg, clear = 0 
 compute_capability = 720, cudnn_half = 0 
net.optimized_memory = 0 
mini_batch = 1, batch = 8, time_steps = 1, train = 1 
   layer   filters  size/strd(dil)      input                output
   0 conv     32       3 x 3/ 1    608 x 608 x   3 ->  608 x 608 x  32 0.639 BF
   1 conv     64       3 x 3/ 2    608 x 608 x  32 ->  304 x 304 x  64 3.407 BF
   2 conv     64       1 x 1/ 1    304 x 304 x  64 ->  304 x 304 x  64 0.757 BF
   3 route  1 		                           ->  304 x 304 x  64 
   4 conv     64       1 x 1/ 1    304 x 304 x  64 ->  304 x 304 x  64 0.757 BF
   5 conv     32       1 x 1/ 1    304 x 304 x  64 ->  304 x 304 x  32 0.379 BF
   6 conv     64       3 x 3/ 1    304 x 304 x  32 ->  304 x 304 x  64 3.407 BF
   7 Shortcut Layer: 4,  wt = 0, wn = 0, outputs: 304 x 304 x  64 0.006 BF
   8 conv     64       1 x 1/ 1    304 x 304 x  64 ->  304 x 304 x  64 0.757 BF
   9 route  8 2 	                           ->  304 x 304 x 128 
  10 conv     64       1 x 1/ 1    304 x 304 x 128 ->  304 x 304 x  64 1.514 BF
  11 conv    128       3 x 3/ 2    304 x 304 x  64 ->  152 x 152 x 128 3.407 BF
  12 conv     64       1 x 1/ 1    152 x 152 x 128 ->  152 x 152 x  64 0.379 BF
  13 route  11 		                           ->  152 x 152 x 128 
  14 conv     64       1 x 1/ 1    152 x 152 x 128 ->  152 x 152 x  64 0.379 BF
  15 conv     64       1 x 1/ 1    152 x 152 x  64 ->  152 x 152 x  64 0.189 BF
[... 140 lines cut ...]
 152 conv    512       3 x 3/ 2     38 x  38 x 256 ->   19 x  19 x 512 0.852 BF
 153 route  152 116 	                           ->   19 x  19 x1024 
 154 conv    512       1 x 1/ 1     19 x  19 x1024 ->   19 x  19 x 512 0.379 BF
 155 conv   1024       3 x 3/ 1     19 x  19 x 512 ->   19 x  19 x1024 3.407 BF
 156 conv    512       1 x 1/ 1     19 x  19 x1024 ->   19 x  19 x 512 0.379 BF
 157 conv   1024       3 x 3/ 1     19 x  19 x 512 ->   19 x  19 x1024 3.407 BF
 158 conv    512       1 x 1/ 1     19 x  19 x1024 ->   19 x  19 x 512 0.379 BF
 159 conv   1024       3 x 3/ 1     19 x  19 x 512 ->   19 x  19 x1024 3.407 BF
 160 conv    255       1 x 1/ 1     19 x  19 x1024 ->   19 x  19 x 255 0.189 BF
 161 yolo
[yolo] params: iou loss: ciou (4), iou_norm: 0.07, cls_norm: 1.00, scale_x_y: 1.05
nms_kind: greedynms (1), beta = 0.600000 
Total BFLOPS 128.459 
avg_outputs = 1068395 
 Allocate additional workspace_size = 106.46 MB 
 Try to load weights: /home/unmnd/catkin_ws/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolov4.weights 
Loading weights from /home/unmnd/catkin_ws/src/darknet_ros/darknet_ros/yolo_network_config/weights/yolov4.weights...
 seen 64, trained: 32032 K-images (500 Kilo-batches_64) 
Done! Loaded 162 layers from weights-file 

 try to allocate additional workspace_size = 106.46 MB 
 CUDA allocate done! 
Waiting for image.

```

Shame-fight, 2020-06-18:
I can run yolov3 at 2-3 frames on jetson-tx2, I have not tried v4. If you are confused about the frame rate, there is a way to verify: run yolov4 (non-ros) alone, run darknet-ros, compare Are the frame rates the same? By the way: v4 has a lower frame rate than v3.
@jorgemia

Hiroaki-K4, 2020-08-06:
I tried building with jetson xavier using darknet_ros (https://github.com/tom13133/darknet_ros) and yolov4, but the following error appears. Is there any solution?
`/home/nvidia/catkin_ws/devel/.private/darknet_ros/lib/libdarknet_ros_lib.so: undefined reference to `show_image_cv_ipl``
`/home/nvidia/catkin_ws/devel/.private/darknet_ros/lib/libdarknet_ros_lib.so: undefined reference to `ipl_into_image``
`/home/nvidia/catkin_ws/devel/.private/darknet_ros/lib/libdarknet_ros_lib.so: undefined reference to `ipl_to_image``

mbjelonic, 2021-04-09:
We will work on yolo v4. :)

closed 2021-06-08 by tomlankhorst