leggedrobotics/darknet_ros #245: Cannot load weights #122

bghadge, 2020-06-04:
I have following weights in darknet_ros/darknet_ros/yolo_network_config/weights
1. yolov2.weights
2. yolov2-tiny.weights
3. yolov3.weights

But darknet cannot find weights. I can't understand where the problem is.

roslaunch darknet_ros/darknet_ros/launch/darknet_ros.launch 

... logging to /home/bhushan/.ros/log/6f8f2620-a66e-11ea-b732-0cb6d2d0d81f/roslaunch-bhushan-Inspiron-5547-4056.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Resource not found: The following package was not found in <arg default="$(find darknet_ros)/yolo_network_config/weights" name="yolo_weights_path"/>: darknet_ros
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

darrenjkt, 2020-06-10:
I faced a similar problem on my first run. The solution for me was simply to add an extra / at the end of the paths in /darknet_ros/darknet_ros/launch/darknet_ros.launch 

i.e. In /darknet_ros/darknet_ros/launch/darknet_ros.launch
```
  <!-- Config and weights folder. -->
  <arg name="yolo_weights_path"          default="$(find darknet_ros)/yolo_network_config/weights/"/>
  <arg name="yolo_config_path"           default="$(find darknet_ros)/yolo_network_config/cfg/"/>
```

tomlankhorst, 2020-06-16:
@bghadge Did $darrenjkt's solution work for you?

bghadge, 2020-06-16:
Thank you @darrenjkt It worked.

closed 2020-06-16 by bghadge

tomlankhorst, 2020-06-17:
This is great to hear. Good luck!