leggedrobotics/darknet_ros #246: Bounding Boxes not being displayed

AndreejS, 2020-06-14:
I am using the yolov3.yaml as the network_param_file. The object detection is working and i also get the percentages shown in the console. However, i do not get any bounding boxes displayed. 

Are there any limitations to the image size ? Our camera sensor creates 128x128 images.

The only thing i found was, that the YoloObjectDetector.hpp has the following errors:
[image]

Also, i can not import the BoundingBox.msg from a different node.
[image]

Can anyone provide some hints on how to solve the problem ?

tomlankhorst, 2020-06-17:
If everything compiles then the includes are found by the compiler. So this might have to do with your IDE configuration. That doesn't explain the bounding boxes issue though. Did you find a solution? 
To pin down whether it's due to your relatively small image size, you could try to feed a large image instead.

AndreejS, 2020-06-22:
The bounding box problem was due to the image resolution the camera sensor was producing. The sensor was producing 128x128 images, we increased that to 500x500 and the bounding boxes were displayed afterwards. Thanks for your hint!

We still did not found the reason for our IDE not recognizing the msg which darknet_ros produces. We are using Vscode, any hints ?

tomlankhorst, 2020-11-28:
That probably has to do with the include paths. You might want to source your workspace in a terminal (e.g. source devel/setup.bash) and load VSCode after that. If it remains a problem feel free to open a new issue.

closed 2020-11-28 by tomlankhorst